project-chip/connectedhomeip #19178: [spec] pump controller app cluster definitions not up to spec

andy31415, 2022-06-03:
#### Problem

```
./scripts/idl_lint.py --log-level warn ./examples/pump-controller-app/pump-controller-common/pump-controller-app.matter
2022-06-03 10:18:34 ERROR   ERROR: Required attributes: EP0:OtaSoftwareUpdateRequestor does not expose DefaultOtaProviders(0) attribute at ./examples/pump-controller-app/pump-controller-common/pump-controller-app.matter:1364:3
2022-06-03 10:18:34 ERROR   Found 1 lint errors
```

ReneJosefsen, 2022-06-19:
@andy31415, seems like the PR was not linked to the issue properly but I think this issue can be closed?

andy31415, 2022-09-14:
Tested today, `./scripts/idl_lint.py --log-level warn ./examples/pump-controller-app/pump-controller-common/pump-controller-app.matter` gives no errors.

closed 2022-09-14 by andy31415