project-chip/connectedhomeip #19180: [spec] lighting app cluster definitions not up to spec

andy31415, 2022-06-03:
#### Problem

```
./scripts/idl_lint.py --log-level warn ./examples/lighting-app/lighting-common/lighting-app.matter
2022-06-03 10:20:02 ERROR   ERROR: Required attributes: EP0:Switch does not expose number of positions(0) attribute at ./examples/lighting-app/lighting-common/lighting-app.matter:1897:3
2022-06-03 10:20:02 ERROR   ERROR: Required attributes: EP0:Switch does not expose current position(1) attribute at ./examples/lighting-app/lighting-common/lighting-app.matter:1897:3
2022-06-03 10:20:02 ERROR   Found 2 lint errors
```

jmartinez-silabs, 2022-06-03:
Switch should probably not be enabled at all for the lighting-app and definitely not on EP0

jepenven-silabs, 2023-02-06:
Wasn't fixed

PR #24977 (2023-02-11): Adding required fields for Light App
Fixing linting errors called out via:

`./scripts/idl_lint.py --log-level warn ./examples/lighting-app/lighting-common/lighting-app.matter`

If this isn't the right way to solve this issue or you'd prefer the switch be removed, let me know. I'm new here :)

Closes #19180
closed 2023-02-11 by andy31415