project-chip/connectedhomeip #19186: [pre-sve] T

doublemis1, 2022-06-03:
#### Problem
all cluster app
nrf52840
revision: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152

Test Harness Client then attempts to write OccupiedCoolingSetpoint below the MinCoolSetpointLimit and above the MaxCoolSetpointLimit and confirms that the device does not accept the value. | Write command should return the error CONSTRAINT_ERROR (0x87)
-- | --

The Device should not accept setting setpoint to out of range value <min, max>.

Min:
```
[1654267783.647256][105075:105080] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0017 DataVersion: 291366426
[1654267783.647309][105075:105080] CHIP:TOO:   MinCoolSetpointLimit: 1600
```
Max:
```
[1654267824.607418][105187:105192] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0018 DataVersion: 291366426
[1654267824.607454][105187:105192] CHIP:TOO:   MaxCoolSetpointLimit: 3200
```

./chip-tool thermostat write occupied-cooling-setpoint 9999 2 1
```
[1654267854.290583][105332:105337] CHIP:DMG: 			StatusIB =
[1654267854.290596][105332:105337] CHIP:DMG: 			{
[1654267854.290608][105332:105337] CHIP:DMG: 				status = 0x00 (SUCCESS),
[1654267854.290618][105332:105337] CHIP:DMG: 			},
```

./chip-tool thermostat write occupied-cooling-setpoint 1000 2 1
```
[1654267878.120926][105399:105404] CHIP:DMG: 			StatusIB =
[1654267878.120937][105399:105404] CHIP:DMG: 			{
[1654267878.120951][105399:105404] CHIP:DMG: 				status = 0x00 (SUCCESS),
[1654267878.120963][105399:105404] CHIP:DMG: 			},
```

andy31415, 2022-06-03:
I am not sure if we should support full min/max functionality in an all-clusters example app. 
Supporting some of these seems to start amounting to "implement a reasonably working mock device". Individually having checks in each of these is probably possible ... but should we commit to something like this for all possible clusters?

bzbarsky-apple, 2022-06-03:
> revision: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152

Old revision.  This was fixed in #18816.

closed 2022-06-03 by bzbarsky-apple