project-chip/connectedhomeip #19191: Cannot pair M5 board (all cluster app) via scanning QR code

phonnakasturi-apple, 2022-06-03:
With SHA 891afbfc0c09445be77eb310a0955bca47f9af02

Cannot pair M5 board (all cluster app) via scanning QR code.

**Steps to reproduce:**

1. cd ~/connectedhomeip
2. git reset --hard 891afbfc0c09445be77eb310a0955bca47f9af02
3. git clean -Xdf
4. git submodule update --init
5. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
6. source scripts/bootstrap.sh
7. source scripts/activate.sh
8. cd ~/tools
9. cd esp-idf
10. git checkout v4.4
11. git submodule update --init --recursive
  ./install.sh
 . ./export.sh
14. cd ~/connectedhomeip/examples/all-clusters-app/esp32
15. idf.py build
16. idf.py menuconfig -> Demo -> choose M5 stack. 

17. idf.py -D 'SDKCONFIG_DEFAULTS=sdkconfig_m5stack.defaults' erase_flash build flash monitor

18. Build chiptool app on iPhone via Xcode

19. Launch chiptool app on iPhone and scan the QR code from M5 board

Attaching the complete M5 board logs:

[M5_Error_logs.txt](https://github.com/project-chip/connectedhomeip/files/8835210/M5_Error_logs.txt)

bzbarsky-apple, 2022-06-04:
@phonnakasturi-apple Is this different from https://github.com/project-chip/connectedhomeip/issues/19135 ?  It looks like the same exact issue to me.  There's no log from the controller side, but the device log shows:
```
I (34900) chip[ZCL]: OpCreds: Received a CSRRequest command
I (36170) chip[ZCL]: OpCreds: NewCertificateSigningRequest succeeded
I (36810) chip[ZCL]: OpCreds: CSRRequest successful.
```
which is longer than 2s to process the command and hence would result in the controller timing out.

This should have been fixed by https://github.com/project-chip/connectedhomeip/pull/19137.  Please retest on a SHA after that merged?

phonnakasturi-apple, 2022-06-07:
Yes this  was fixed with  SHA dd636ad0d5053e13341733c8c6205d9a98c1846c
I will close this

closed 2022-06-07 by phonnakasturi-apple