project-chip/connectedhomeip #19192: [pre-sve] NotifyUpdateApplied fails if triggered before the node joins the fabric

selissia, 2022-06-03:
#### Problem
When a node reboots after a successful OTA update the sending NotifyUpdateApplied might get triggered before the Node re-joins the fabric. In this case the command fails to be transmitted. This fails the TC-SU-2.6 test case

For example, in case of Thread transport re-joining the Thread network takes some time, see attached log snippet.

 #### Proposed Solution
OTARequestor class should delay sending NotifyUpdateApplied until the device has joined the fabric.

selissia, 2022-06-03:
[
[tc-su-26-dut-log.txt](https://github.com/project-chip/connectedhomeip/files/8835432/tc-su-26-dut-log.txt)
](url)

carol-apple, 2022-06-04:
I think it'd be tricky to figure out how long to delay. There are no timestamps in your log snippet. How long between the first failed `Advertise operational node 2BF32B802FF0AE27-0000000000000002` vs the second one which was successful?

selissia, 2022-06-04:
Several seconds, I don't have a more specific info than that. But this will vary between devices and network conditions anyway. 
I meant "delay" in a more general sense, as in "don't attempt NotifyUpdateApplied until the device has joined the fabric" -- how exactly to accomplish that I'm not yet sure.

marius-alex-tache, 2022-09-20:
Hi! I'm seeing this after trying to increase OT baud rate to 921600. Seems like the call to `HandleDnssdInit` is delayed quite a lot. The only thing that got it to work was a combination between a timer and capturing the `kDnssdPlatformInitialized` event.

closed 2023-04-02 by stale[bot]

eme-bennytang, 2023-10-24:
Hi all,

I think I am seeing this problem here, I found another issue that is related but my code has already implemented that change
https://github.com/project-chip/connectedhomeip/pull/22131

Is there any further update for the OTA state machine for this?