project-chip/connectedhomeip #19194: Darwin DNS-SD implementation stops browsing services upon finding a wrong service.

kpark-apple, 2022-06-03:
#### Problem
Darwin chip::Dnssd::Browse() stops browsing when OnBrowse() was called without kDNSServiceFlagsMoreComing flag.
Per https://developer.apple.com/documentation/dnssd/1823436-anonymous/kdnsserviceflagsmorecoming?language=objc, the flag only means that there are no more "queued" results.
The issue with it is that if the desired service is discovered later than an undesirable service (service from another device for example), the desired service is never discovered within the timeout, causing failure.
 
 #### Proposed Solution
Remove sdCtx->Finalize() call with the flag check, and make sure the context is finalized when desired service was found or when timeout occurs.

bzbarsky-apple, 2022-06-04:
There is no timeout on browse right now, fwiw...

I guess since we switched to browsing for the _CM subtype there's a problem if multiple commissionable things are all advertising because the discriminator filter is applied too late?  Is that why this is being a problem?  How often does this behavior manifest as an issue?

The proposed solution on its own is not enough.  The caller of Browse can only handle a single callback, so we would need to either change that or always keep going until the timeout and deliver all the results we found, which would leave to pretty undesirable behavior.  And the caller is the generic platform mdns code, so changing that would involve changing all the other platform implementations too...

kpark-apple, 2022-06-04:
> I guess since we switched to browsing for the _CM subtype there's a problem if multiple commissionable things are all advertising because the discriminator filter is applied too late? Is that why this is being a problem? How often does this behavior manifest as an issue?
Yes, that is the problem and it happened very frequently with test team. It may be unique in test setups where they might have some other commissionable node running. However, personally I would think it could happen frequently in real world, too, if the commissioning window is as long as a minute or so.

PR #22823 (2022-10-04): Add an API to stop a DNS-SD browse operation.
Most backends don't implement this yet. Darwin does, and no longer stops Browse operations itself.

May provide a way toward fixing
https://github.com/project-chip/connectedhomeip/issues/13275

#### Issue Being Resolved
* Fixes #19194

#### Change overview
1. Add `ChipDnssdStopBrowse`
2. Don't stop browse operations on Darwin until `ChipDnssdStopBrowse` is called.
3. Add the relevant `ChipDnssdStopBrowse` call to `SetUpCodePairer`.
closed 2022-10-04 by andy31415