project-chip/connectedhomeip #19213: [pre-sve] Spec Limits are no longer being enforced

chrisbeach59, 2022-06-06:
#### Problem
Step 2b of TSTST-2.2 is failing
 * expected behavior
 * Setting a attribute outside of limits should fail
 * actual behavior
 No limits are enforced
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
This used to work.   It used the global pre-attribute-change call back and looked for the thermostat cluster.  There was a PR that was supposed to transition it to use the cluster specific pre-attribute change callback.  This does not seem to work.

bzbarsky-apple, 2022-06-07:
@chrisbeach59 We have CI verifying that the limits are enforced...  Which exact SHA are you using, which client, which server app?

bzbarsky-apple, 2022-06-07:
I just tried current tip (55ab764beabaed8675bd60e6d18e84d545a733e0), "linux" all-clusters-app commissioned as node id 17, running the following chip-tool commands:

1. Read the limits
    ```
    chip-tool thermostat read min-cool-setpoint-limit 17 1
    chip-tool thermostat read max-cool-setpoint-limit 17 1
   ```
   I get 1600 and 3200 respectively.
2. Try to set the setpoint:
     ```
     chip-tool thermostat write occupied-cooling-setpoint 1400 17 1
     chip-tool thermostat write occupied-cooling-setpoint 3300 17 1
     ```
     and both get CONSTRAINT_ERROR.

I also tried the same thing with the "linux" thermostat app, and I get the same results.

chrisbeach59, 2022-06-07:
@bzbarsky-apple am able to validate that I see the issue in the linux build for the pre-sve week 1 sha.  And also that if I pull ToT the problem is gone. Hopefully the week 2 sha will include this fix and I will close.

bzbarsky-apple, 2022-06-07:
@chrisbeach59 Of which app?  The pre-sve week 1 SHA likely does not include the fix for #18816 so if you're using all-clusters-app it's expected to fail there.

chrisbeach59, 2022-06-07:
Thermostat app.  So expected to fail in pre-sve

bzbarsky-apple, 2022-06-08:
Thermostat had checks already, though via https://github.com/project-chip/connectedhomeip/pull/18816/files#diff-cc5b98517f74e271ae77706ffd8241fec62bf8bf939e87de47451284ac037107L48

So I'm not sure why that would be failing.

chrisbeach59, 2022-06-08:
My original implementation was before cluster specific pre attribute
changed callbacks were in the sdk.  I used global callback then checked for
thermostat cluster.  I think the Sha is somehow in-between.

andy31415, 2022-06-13:
@chrisbeach59 - is this fixed in ToT?

I would like to close it if it is gone in master (Boris seems to indicate that this may work on 55ab764).

andy31415, 2022-06-13:
Closing based on comment of `if I pull ToT the problem is gone`

closed 2022-06-13 by andy31415

chrisbeach59, 2022-10-11:
The week 1 pre-sve sha.
Running on our mattrr development platform board.