project-chip/connectedhomeip #19214: [pre-sve] TC-SU-2.3 fail to start new CASE session with ota-provider

pankore, 2022-06-06:
#### Problem
Test case ID: TC-SU-2.3
SHA: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152
Platform: Ameba all-clusters-app
Chip-tool platform: Raspberry Pi
OTA provider platform: Linux ota-provider on rpi4

- After killing the ota-provider and waiting for the update state to become idle, DUT (ota-requestor) is unable to start CASE session with OTA-provider.
- Referring to https://github.com/project-chip/connectedhomeip/pull/18642, after DUT times out, `[SWU]CASE session may be invalid, tear down session.` is not printed in the DUT logs

Steps to reproduce:
- Commission DUT and OTAP, start OTA transfer by announce-ota-provider command
- Kill OTA image transfer midway, wait for DUT to time out and check its update state == 1 (idle)
- Relaunch a new OTAP session, commission new OTAP
- Start OTA transfer by annouce-ota-provider command

[TC-SU-2.3-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8841871/TC-SU-2.3-chip-tool.txt)
[TC-SU-2.3-DUT.txt](https://github.com/project-chip/connectedhomeip/files/8841872/TC-SU-2.3-DUT.txt)

isiu-apple, 2022-06-06:
In `TC-SU-2.3-DUT.txt`, I do see: `chip[SWU] CASE session may be invalid, tear down session`.  I also see: `chip[SWU] Query already in progress`.  

Are you testing with this change also?  https://github.com/project-chip/connectedhomeip/pull/18765

pankore, 2022-06-07:
@isiu-apple yes I am testing with that change. The `chip[SWU] CASE session may be invalid, tear down session` is not printed out when the initial OTA transfer timeout, but it is printed when it is trying to establish a new session with the new OTA provider. Is this an expected behaviour?

pankore, 2022-06-07:
@isiu-apple no, #18765 was not included. `chip[SWU] CASE session may be invalid, tear down session. I also see: chip[SWU] Query already in progress` is not printed when the initial OTA transfer time out, but is printed when it is querying the new OTA provider.

pankore, 2022-06-07:
Closing this issue as #18765 was not merged at the time of testing. Will test again with SHA that includes the fix and reopen if issue persists. Thanks.

closed 2022-06-07 by pankore