project-chip/connectedhomeip #19228: [Pre-SVE] TC-SWTCH-1.1 DUT has neither LS nor MS feature according to feature map

HubertHugo, 2022-06-06:
all cluster app
nRF52840
revision: 425f1fb916ca154deeb75cd82cd5196778aba152

DUT has set featureMap 0 which means that it doesn't support neither LS nor MS feature:

```
[1654522792.126108][20440:20445] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_003B Attribute 0x0000_FFFC DataVersion: 1019132376
[1654522792.126152][20440:20445] CHIP:TOO:   FeatureMap: 0
```

Yet attribute list suggests that it supports MSM:

```
[1654522783.917191][20433:20438] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_003B Attribute 0x0000_FFFB DataVersion: 1019132376
[1654522783.917289][20433:20438] CHIP:TOO:   AttributeList: 8 entries
[1654522783.917316][20433:20438] CHIP:TOO:     [1]: 0
[1654522783.917329][20433:20438] CHIP:TOO:     [2]: 1
[1654522783.917340][20433:20438] CHIP:TOO:     [3]: 2
[1654522783.917351][20433:20438] CHIP:TOO:     [4]: 65528
[1654522783.917362][20433:20438] CHIP:TOO:     [5]: 65529
[1654522783.917374][20433:20438] CHIP:TOO:     [6]: 65531
[1654522783.917383][20433:20438] CHIP:TOO:     [7]: 65532
[1654522783.917399][20433:20438] CHIP:TOO:     [8]: 65533
```

woody-apple, 2022-06-07:
Cert Blocker Review: Assigning to @bzbarsky-apple to put up simple examples here

woody-apple, 2022-06-07:
Cert Blocker Review: We don't believe this is a cert blocker, given the is implementable and testable given the current SDK. That said, some examples could be helpful here.

bzbarsky-apple, 2022-06-08:
Doing an example with both LS and MS in the same app is a bit of a pain, because then you have to have a useful Fixed Label cluster and our current Fixed Label setup is pretty broken.

For now, just going to set the one in all-clusters-app to be LS, like all-clusters-minimal-app.

closed 2022-06-10 by bzbarsky-apple