project-chip/connectedhomeip #19234: ESP32-C3 initial communication issue

agners, 2022-06-06:
#### Problem
Using the CHIP Python library trying to communicate to a ESP32-C3 running the lightning example right after commissioning seems to consistently fail with:

```
2022-06-06 16:04:06 allenwind chip.DIS[52691] ERROR Operational discovery failed for 0x00000000000010F0: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
```

Trying to reproduce using `chip-tool` I get the same error, although from a slightly different location. I think the reason for the difference is because I run a new `chip-tool` instance to execute the command. The underlying culprit is probably the same.

```
[1654522644.493970][47135:47135] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:408: CHIP Error 0x00000032: Timeout
```

[esp32-c3.log](https://github.com/project-chip/connectedhomeip/files/8845178/esp32-c3.log)

[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8845180/chip-tool.log)

We have observed the same behavior in at least two setups, so it seems not limited to my environment.

 #### Proposed Solution
None yet

agners, 2022-06-06:
Using the interactive mode I am able to reproduce the issue with the same error location as in our Python based implementation.

[esp32-c3-second.log](https://github.com/project-chip/connectedhomeip/files/8845407/esp32-c3-second.log)
[chip-tool-second.log](https://github.com/project-chip/connectedhomeip/files/8845408/chip-tool-second.log)

From what I can tell, the initial operational node discovery using mDNS works as expected. However, the initial Matter communication fails for some reason:

```
[1654527790.702790][72809:72822] CHIP:EM: Failed to Send CHIP MessageCounter:16777679 on exchange 36391i sendCount: 4 max retries: 4
[1654527794.702836][72809:72822] CHIP:DIS: MINMDNS: Sending query ANY/MULTICAST for E7421DA31E57D67C-000000000000153B._matter._tcp.local
[1654527802.705312][72809:72822] CHIP:DIS: Checking node lookup status after 15006 ms
[1654527802.705335][72809:72822] CHIP:DIS: Operational discovery failed for 0x000000000000153B: ../../examples/chip-tool/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
```

agners, 2022-06-07:
This actually looks an awful lot like #19135. I'll retest tomorrow with the latest version.

agners, 2022-06-07:
I just tested master branch of Matter, but the problem still persists, so it seems #19135 is not solving it.

@dhrishi thanks for confirming! I'll try with that option next.

agners, 2022-06-07:
@dhrishi just tried firmware with disabled `CONFIG_DEINIT_BLE_ON_COMMISSIONING_COMPLETE`, and it doesn't show the network drop anymore. Reading attributes right after onboarding works fine now. I'll keep this work around in place for now until the upgrade to v4.4.2 comes in. Thanks for your help.

dhrishi, 2022-06-07:
@agners Great! Glad to know it works fine. I will close this issue now.

closed 2022-06-07 by dhrishi

dhrishi, 2022-08-19:
> Once v4.4.2 is out, we will update Matter to use that.

@agners ESP-IDF for Matter is now updated to the version v4.4.2. With this and the latest `connectedhomeip` codebase you can enable `CONFIG_DEINIT_BLE_ON_COMMISSIONING_COMPLETE` and should not see this issue. I don't see it on ESP32-C3. Please give it a try.

tonywu-uec, 2022-11-15:
> @agners ESP-IDF for Matter is now updated to the version v4.4.2. With this and the latest `connectedhomeip` codebase you can enable `CONFIG_DEINIT_BLE_ON_COMMISSIONING_COMPLETE` and should not see [...]

I am using examples/lighting-app/esp32 and could not find the CONFIG_DEINIT_BLE_ON_COMMISSIONING_COMPLETE setting in the sdkconfig file. Can you tell me where to change it? I still can't make this example commissioning successfully on
my ESP32 C3 board.