project-chip/connectedhomeip #19253: Failed certificate chain validation for Node cert

cwangAmazon, 2022-06-06:
#### Problem
AWS PCAS team is working with us to support CA signing certificates for chip project. We have signed Root cert, ICA cert and Node cert, we can verify they are valid by using Openssl command. However, if we use chip cert tool to do validation, it error out.

Below command/result that we run locally to validate certs are valid by openssl
$ openssl verify -verbose -CAfile <(cat root_ca_cert.pem intermediate_ca_cert.pem) leaf_cert.pem                      
leaf_cert.pem: OK
$ openssl verify -verbose -CAfile <(cat root_ca_cert.pem) intermediate_ca_cert.pem 
intermediate_ca_cert.pem: OK
$ openssl verify -verbose -CAfile <(cat root_ca_cert.pem) root_ca_cert.pem 
root_ca_cert.pem: OK

When we use chip cert tool listed [here](https://github.com/project-chip/connectedhomeip/blob/d7bcb65c806f601c0c47dc446d4979ee86f76414/src/tools/chip-cert/README.md): 

we saw below errors:
$ ./out/host/chip-cert validate-cert /leaf_cert.pem -c /intermediate_ca_cert.pem -t /root_ca_cert.pem
Failed certificate chain validation: ../../src/crypto/CHIPCryptoPALOpenSSL.cpp:740: Error 0x00000014

I can append my certs below.
My question would be, how can this happen? does chip cert tool do sth else to validate cert other than what Openssl command do?

Another side question: if the chip tool passed validation here, can we safely assume the certs are valid? Will the SDK doing same procedure of validation just like what that tool is doing?

Root cert:
-----BEGIN CERTIFICATE-----
MIIBpjCCAUygAwIBAgIRAL+oOQMGKqga/d3evSFkCekwCgYIKoZIzj0EAwIwIjEg
MB4GCisGAQQBgqJ8AQQMEENBQ0FDQUNBMDAwMDAwMDEwHhcNMjIwNjA2MTUxNjUx
WhcNMzIwNjAzMTYxNjUxWjAiMSAwHgYKKwYBBAGConwBBAwQQ0FDQUNBQ0EwMDAw
MDAwMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABJjlpX1Q5yGcFuD/yQ0XVFnA
/9/7iqWJcQxHcZCI9IxNwtzJ07zK/COtECavuXmUnhOzbEuM1AZAxE31Td7v2DSj
YzBhMA8GA1UdEwEB/wQFMAMBAf8wHwYDVR0jBBgwFoAUyoOCwNk7swFMHHDjt01o
R4zKssIwHQYDVR0OBBYEFMqDgsDZO7MBTBxw47dNaEeMyrLCMA4GA1UdDwEB/wQE
AwIBhjAKBggqhkjOPQQDAgNIADBFAiBssKV8Aq/WzUuwgGtUmG4ku00voIdbyCPD
u9jHsq4AGgIhAO1RWOXXw+LdX+qKaYFQA2NsVUf8enFA6yOjKDfgOYMt
-----END CERTIFICATE-----

ICA cert:
-----BEGIN CERTIFICATE-----
MIIBpzCCAU6gAwIBAgIQQed0OQ5Uomov6cFBDm7+lzAKBggqhkjOPQQDAjAiMSAw
HgYKKwYBBAGConwBBAwQQ0FDQUNBQ0EwMDAwMDAwMTAeFw0yMjA2MDYxNTQ2MTFa
Fw0yNTExMDcxNjQ2MTFaMCIxIDAeBgorBgEEAYKifAEDDBBDQUNBQ0FDQTAwMDAw
MDAzMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEj0fT/bnLtC8buh6loxHaEDpo
uYzXEvvRZJK8qLz0uZPvE/Muz/zrG5pWRwcz7Sg6qpLDZqNEzH20VTFJ5U/BVaNm
MGQwEgYDVR0TAQH/BAgwBgEB/wIBADAfBgNVHSMEGDAWgBTKg4LA2TuzAUwccOO3
TWhHjMqywjAdBgNVHQ4EFgQUT+QFQp3O/0rpRixdj7SYD2oldaAwDgYDVR0PAQH/
BAQDAgGGMAoGCCqGSM49BAMCA0cAMEQCIHD0RmwFxFGhRkeUFQdPw+2sglheBwkq
htPIU3eWplWTAiBBPMxIq3kwvRlKAN+tp1VSriVC+1SRRqsNxdPwSC19kQ==
-----END CERTIFICATE-----

Node cert:
-----BEGIN CERTIFICATE-----
MIIB6DCCAY+gAwIBAgIQKyY/fAib3tPtVi32iyFUZjAKBggqhkjOPQQDAjAiMSAw
HgYKKwYBBAGConwBAwwQQ0FDQUNBQ0EwMDAwMDAwMzAeFw0yMjA2MDYxNTU3MzNa
Fw0yMzA2MDYxNjU3MzNaMEQxIDAeBgorBgEEAYKifAEBDBBERURFREVERTAwMDEw
MDAxMSAwHgYKKwYBBAGConwBBQwQRkFCMDAwMDAwMDAwMDAxRDBZMBMGByqGSM49
AgEGCCqGSM49AwEHA0IABHpZMYCGDEA3yDwSdJhFyO4UJN0pf63LiV41glXSx9Ky
qMolWA8mJv5XkGL/G5n/kcJKDaBvsytb4gFIySSfVlCjgYQwgYEwDAYDVR0TAQH/
BAIwADAfBgNVHSMEGDAWgBRP5AVCnc7/SulGLF2PtJgPaiV1oDAdBgNVHQ4EFgQU
IzAdEMBlZn2ZLCo+mqXhu1EE9ZEwDwYDVR0PAQH/BAUDAweAADAgBgNVHSUBAf8E
FjAUBggrBgEFBQcDAQYIKwYBBQUHAwIwCgYIKoZIzj0EAwIDRwAwRAIgaxXTdxPi
YgEys5o4I4g0U5rZTsD1g7zaqPnBYuJSnxACIH/G6kvnbjCAY2ENegS5kGqO7SgV
nWoj1F1QimUoEboT
-----END CERTIFICATE-----

bzbarsky-apple, 2022-06-07:
Comparing the command above:
```
 $ ./out/host/chip-cert validate-cert /leaf_cert.pem -c /intermediate_ca_cert.pem -t /root_ca_cert.pem
```

to the command from the chip-cert README:
```
./chip-cert validate-cert Chip-Node-Cert.chip-b64 -c Chip-ICA-Cert.pem -t Chip-Root-Cert.pem
```
it looks like the node cert is being provided in the wrong format: it's expected to be base64-encoded Matter TLV format, not PEM...

tcarmelveilleux, 2022-06-07:
> Another side question: if the chip tool passed validation here, can we safely assume the certs are valid? Will the SDK doing same procedure of validation just like what that tool is doing?

The chip tooling in SDK all uses the same code.

Regarding the failure, I think the tool is improperly documented for the purpose you are attempting to use it for. Let us look for what's wrong. 

Attached are the PEMs as a ZIP to help others: [cert_candidates.zip](https://github.com/project-chip/connectedhomeip/files/8853422/cert_candidates.zip)

tcarmelveilleux, 2022-06-07:
Overall, the tool does the following on Validate Cert:

- Load the cert, auto detecting format. and always going via X.509 format
  - DER and PEM are loaded, then internally converted to CHIP certificate TLV format
  - base64 and TLV cert are converted to X.509, then to chip certs
- Run validation

The certificates here are failing signature verification:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, chip::Crypto::P256PublicKey::ECDSA_validate_hash_signature (this=0x7fffffffb440, 
    hash=0x5555555ac108 "\002k\304[\245u\256\337<\361ǂ{^`\350\025a\fY", hash_length=32, signature=...)
    at ../../../../../src/crypto/CHIPCryptoPALOpenSSL.cpp:736
warning: Source file is more recent than executable.
736	    result = ECDSA_SIG_set0(ec_sig, r, s);
(gdb) bt
#0  chip::Crypto::P256PublicKey::ECDSA_validate_hash_signature (this=0x7fffffffb440, 
    hash=0x5555555ac108 "\002k\304[\245u\256\337<\361ǂ{^`\350\025a\fY", hash_length=32, signature=...)
    at ../../../../../src/crypto/CHIPCryptoPALOpenSSL.cpp:736
#1  0x000055555556b4ef in chip::Credentials::ChipCertificateSet::VerifySignature (cert=0x5555555abf80, caCert=0x5555555ac128)
    at ../../../../../src/credentials/CHIPCert.cpp:302
#2  0x000055555556be87 in chip::Credentials::ChipCertificateSet::ValidateCert (this=0x7fffffffd5c0, cert=0x5555555abf80, context=..., 
    validateFlags=..., depth=1 '\001') at ../../../../../src/credentials/CHIPCert.cpp:422
#3  0x000055555556c0b6 in chip::Credentials::ChipCertificateSet::FindValidCert (this=0x7fffffffd5c0, subjectDN=..., subjectKeyId=..., 
    context=..., validateFlags=..., depth=1 '\001', certData=0x7fffffffb838) at ../../../../../src/credentials/CHIPCert.cpp:464
#4  0x000055555556bdd8 in chip::Credentials::ChipCertificateSet::ValidateCert (this=0x7fffffffd5c0, cert=0x5555555ac2d0, context=..., 
    validateFlags=..., depth=0 '\000') at ../../../../../src/credentials/CHIPCert.cpp:414
#5  0x000055555556c0b6 in chip::Credentials::ChipCertificateSet::FindValidCert (this=0x7fffffffd5c0, subjectDN=..., subjectKeyId=..., 
    context=..., validateFlags=..., depth=0 '\000', certData=0x7fffffffd5b8) at ../../../../../src/credentials/CHIPCert.cpp:464
#6  0x000055555556b30a in chip::Credentials::ChipCertificateSet::FindValidCert (this=0x7fffffffd5c0, subjectDN=..., subjectKeyId=..., 
    context=..., certData=0x7fffffffd5b8) at ../../../../../src/credentials/CHIPCert.cpp:284
#7  0x0000555555566ec4 in Cmd_ValidateCert (argc=6, argv=0x7fffffffd7f0) at ../../../../../src/tools/chip-cert/Cmd_ValidateCert.cpp:188
#8  0x000055555556a3a6 in main (argc=7, argv=0x7fffffffd7e8) at ../../../../../src/tools/chip-cert/chip-cert.cpp:122
(gdb) quit

```

Since they passed the Convert step, they are not initially wrong.

emargolis, 2022-06-14:
@cwangAmazon Could you please update the status of this issue. 

Where you able to identify the problem and fix it?

chandarana, 2022-06-14:
We were able to fix cert and CHIP tool validation passed.

We did run into some other issues when actually connecting devices. We are investigating that issue. Most likely, it is not related to op-cert.

ChetanSaiKumarThalisetty, 2022-06-14:
Well we were able to remove the extra bits in KeyUsage and make the validations succeed. But Since openssl is not having problem with these extra bits. Should we update validation of chip-cert to make it inline with openssl ?

emargolis, 2022-08-10:
@ChetanSaiKumarThalisetty I believe this specific issue was resolved and I am closing this ticket

closed 2022-08-10 by emargolis