project-chip/connectedhomeip #19259: Make `OperationalDeviceProxy` shareable by multiple applications, and delivers its callbacks to multiple applications

kghost, 2022-06-07:
#### Problem
* 1: If an application context is released before the pairing completes, will render the pointer passed into `OnDeviceConnected` dangling
* 2: in `OperationalDeviceProxy::DequeueConnectionCallbacks` we may use member fields which is already destructed
  * `cb->mCall(cb->mContext, mPeerId, error)`
    * We are loop for all callbacks here, but `this` object may already destructed by previous callback, so `mPeerId` may not valid here
  * `cb->mCall(cb->mContext, this)`
    * Obviously, `this` may not valid here

 #### Proposed Solution
* Implement a type-safe and RAII style callback queue using `IntrusiveList` to solve problem 1
* When implementing `DequeueConnectionCallbacks`, split the function into 2 parts:
  1. Move the callback queue to a stack variable
  2. Use an static member function to call all callback, to ensure that no member fields is used

bzbarsky-apple, 2022-06-07:
The plan is to stop passing `this` to the callbacks altogether and just pass the session handle, right?  So problem 1 is not worth worrying about, I suspect.

The use of `mPeerId` is a problem; very good catch there.  Factoring out the callback-calling into a static function to avoid those problems seems good.

kghost, 2022-06-07:
In `DeviceCommissioner::OnDeviceConnectedFn`, `this` (an `OperationalDeviceProxy`) is being used. So, problem 1 still need some attention.

bzbarsky-apple, 2022-06-08:
> The context pointer in the callback is dangling, calling the callback after its context is released will yield pretty bad result.

Anything that releases the context should Cancel() the callback, no?  That's why we are using cancelables here....  I thought you meant the pointer to the OperationalDeviceProxy could dangle, for point 1.

> In DeviceCommissioner::OnDeviceConnectedFn, this (an OperationalDeviceProxy) is being used. So, problem 2.2 still need some attention.

Problem 2.2 is what "just pass in the session instead" will solve, right?

> I'm really frustrated dealing with object freeing itself inside its member function

Sure.

> can we just give OperationalDeviceProxy a reference counter and call it a day ?

Again, the plan is to not have OperationalDeviceProxy _at all_.

kghost, 2022-06-15:
@bzbarsky-apple When trying to solve problem 2.2, the `OperationalDeviceProxy` object is passed to `ClusterBase::Associate` which assigned to `ClusterBase::mDevice`.

So I have to change `mDevice` to a session handle, this should be a straightforward and simple change, but not small. Also need to add a member of pointer to `ExchangeManager` to `ClusterBase`, which may impact memory usage.

mrjerryjohns, 2022-06-15:
I'm labelling this 1.0 since this will become more imperative when we have multiple consumers of `OperationalDeviceProxy` in the SDK (IM will be one of them...)

bzbarsky-apple, 2022-06-16:
@kghost Just to allay the ClusterBase concern: those are stack-only objects in practice.

mrjerryjohns, 2022-07-07:
This is not completed. #19673 just unblocks the core work for this issue.

mrjerryjohns, 2022-07-07:
As part of addressing this, we should also remove `CASESessionManager::Release*` APIs as well.

woody-apple, 2022-08-03:
SVE/Cert Blocker Review: Does not appear to be blocking a test case, removing SVE.

PR #19673 (2022-07-06): Remove DeviceProxy pointer in ClusterBase
#### Problem
Pave the path to fix #19259

#### Change overview
Remove `DeviceProxy*`  in `ClusterBase` class, use a session handle ref instead.

~~@andy31415 Not sure how to generate files under `scripts/idl/tests/outputs`~~
@bzbarsky-apple Need to fix `src/darwin/Framework/CHIP/CHIPCluster.mm` but I don't understand object-c at all.

#### Testing
Refactor, can be verified by unit tests.
PR #21256 (2022-08-05): No longer provide OperationDeviceProxy in OnDeviceConnected callback
#### Problem
What is being fixed?  Examples:
* Fixes #19259
* Previous implementations of OnDeviceConnected held onto OperationalDeviceProxy when they really should not have could lead to use after free should something else free that OperationalDeviceProxy.

#### Change overview
* New API makes it more clear that the SessionHandle is intended to be consumed in the callback itself.
* Since many current application incorrectly implemented OnDeviceConnected, we allow application to create create and manage the lifetime of their [... cut]
closed 2022-08-05 by tehampson