project-chip/connectedhomeip #19265: [pre-sve] TC-LVL-4.1 CurrentLevel does not reach MinLevel

pankore, 2022-06-07:
#### Problem
TC-LVL-4.1
Platform: Ameba all-clusters-app

- According to specs, the CurrentLevel ranges from MinLevel to MaxLevel.
- DUT reads MinLevel = 0, MaxLevel = 254.
- When TH sends move downwards command, CurrentLevel stops at 1 instead of 0

[TC-LVL-4.1.txt](https://github.com/project-chip/connectedhomeip/files/8852151/TC-LVL-4.1.txt)

bzbarsky-apple, 2022-06-07:
@pankore Does the app claim the LT feature is supported or not?

If LT is supported, MinLevel 0 is not valid; it's defined to be "1 to MaxLevel" if LT is supported.

@jmartinez-silabs

jmartinez-silabs, 2022-06-07:
Level control featuremap in all-app cluster is set to the value 3

```
            {
              "name": "FeatureMap",
              "code": 65532,
              "mfgCode": null,
              "side": "server",
              "type": "bitmap32",
              "included": 1,
              "storageOption": "RAM",
              "singleton": 0,
              "bounded": 0,
              "defaultValue": "3",
              "reportable": 1,
              "minInterval": 1,
              "maxInterval": 65534,
              "reportableChange": 0
            },
```
[image]

So OnOff and Lighting features. So if current level stops at 1 then this is the expected behaviour

bzbarsky-apple, 2022-06-08:
> So OnOff and Lighting features. So if current level stops at 1 then this is the expected behaviour

Yes, the unexpected part is that MinLevel returns 0.

closed 2022-06-08 by jmartinez-silabs