project-chip/connectedhomeip #19272: [pre-sve] TC-MOD-3.3  The mode backs to the start-up-mode value after matter ota.

doublemis1, 2022-06-07:
#### Problem
all cluster app
nrf52840
revision: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152

1 | 1.8.5 | MOD.S.A0004 | TH reads the StartUpMode attribute from the DUT | Verify that the DUT response is an integer. Record this value for usage in step 2.
-- | -- | -- | -- | --
2a | 1.8.5 | MOD.S.A0004 | TH reads the CurrentMode attribute from the DUT | Verify that the DUT response is an integer. Record this value for usage in steps 2b and 3.
2b |   | MOD.S.A0004 | If the StartUpMode and CurrentMode attributes have the same value, proceed to step 2c, Otherwise proceed to step 3a. |  
2c | 1.8.5, 1.8.9 | MOD.S.A0004 | TH reads the SupportedModes attribute from the DUT | Verify that the DUT response provides a list of supported modes. Record the mode numbers for usage in step 2d.
2d | 1.8.6 | MOD.S.A0004 | Select a value from the list in step 2c with a different value than the StartUpMode value read in step 1. TH sends a ChangeToMode command to the DUT with the selected value. | Verify DUT responds with a successful (value 0x00) status response. Record the value provided for usage in step 3.
3a |   | MOD.S.A0004 | Perform an OTA update on the device that requires a reboot. Allow for the DUT to update and fully reboot. |  
3b | 1.8.5 | MOD.S.A0004 | TH reads the CurrentMode attribute from the DUT | Verify that the DUT response is the integer recorded in step 2a or, if the mode was changed, 2d.

Step3a failed. The mode backs to the start-up-mode value after ota.

 #### Discussion
t It can be also reproduced by DUT reboot (not OTA).

woody-apple, 2022-07-20:
Cert Blocker Review: Naveen to follow up about this test case.

bzbarsky-apple, 2022-07-21:
For what it's worth, the testcase looks correct to me....

closed 2022-07-23 by woody-apple