project-chip/connectedhomeip #19276: MF-10 and MF-1.9 tests failes for Darwin(no-ble-asan) showing Invalid PASE parameter

manjunath-grl, 2022-06-07:
Summary:
As per testplan TC-MF-1.10 TH start attempting to do PASE to DUT_CE and failing 20 times for Invalid payload value but  Test Suites - Darwin (no-ble-asan) seems to be failing showing that Invalid PASE Parameter

Execution logs:
[image]
[image]
[logs_1075704.zip](https://github.com/project-chip/connectedhomeip/files/8854977/logs_1075704.zip)

manjunath-grl, 2022-06-07:
> @manjunath-grl So this is for darwin-framework-tool and specifically for using `PairWithCode` with an incorrect code?

@bzbarsky-apple yes

manjunath-grl, 2022-06-07:
> @manjunath-grl I can't reproduce this.  I pulled the code for PR https://github.com/project-chip/connectedhomeip/pull/18521 and ran:
> ```
> ./scripts/run_in_build_env.sh "./scripts/tests/run_test_suite.py --chip-tool ./out/debug/standalone/darwin-framework-tool --target Test_TC_MF_1_10 run --iterations 1 --all-clusters-app ./out/debug/standalone/chip-all-clusters-app"
> ```
> and it passed fine for me....

I disabled the tests MF-1.10 and MF-1.9 from Darwin tests.js as per @raju-apple request can you please enable them and test it.

bzbarsky-apple, 2022-06-07:
I pulled the PR before you did that, so I don't have the test disabled.  It just passes.

bzbarsky-apple, 2022-06-07:
@manjunath-grl As in the CI failed?

bzbarsky-apple, 2022-06-07:
Note that the bits about "invalid PASE parameter" are absolutely expected.  What's not necessarily expected is a OnPairingComplete call with error after that.

bzbarsky-apple, 2022-06-07:
@manjunath-grl I tried taking your PR and re-enabling the Darwin tests, then running CI and I do see it fail.  I wonder why I can't reproduce that locally...

bzbarsky-apple, 2022-06-07:
Ah, interesting, that failure is in MF-1.9, not MF-1.10.  All the text above claimed the failure was in MF-1.10.

But locally I can't reproduce the failure in MF-1.9 either.

bzbarsky-apple, 2022-06-08:
Still can't reproduce this locally, but the linked PR fixes it in CI...

closed 2022-06-08 by andy31415