project-chip/connectedhomeip #19304: In cc13x2x7_26x2x7, LogEvent seems to emanate from a thread other than the CHIP task.

yunhanw-google, 2022-06-08:
I think `cc13x2x7_26x2x7`'s main.cpp contains a call to `LogEvent` that seems to emanate from a thread other than the CHIP task.

Might want to file an issue on that.

_Originally posted by @mrjerryjohns in https://github.com/project-chip/connectedhomeip/pull/19285#pullrequestreview-998905813_

yunhanw-google, 2022-06-08:
@andreilitvin  who is taking care of cc13x2x7_26x2x7?

andy31415, 2022-06-08:
> @andreilitvin who is taking care of cc13x2x7_26x2x7?

@srickardti maybe based on commit logs?

srickardti, 2022-06-08:
@yunhanw-google can you point me to the offending code? The only reference to logging in our `main.cpp` files is a commented out platform log because the UART driver isn't up until the kernel is started.

In our examples we usually use a macro to call our platform logging function. Is there a better way to consolidate logging?

bzbarsky-apple, 2022-06-08:
```
connectedhomeip % git grep -n LogEvent | grep cc13
examples/pump-app/cc13x2x7_26x2x7/main/AppTask.cpp:621:        if (CHIP_NO_ERROR != LogEvent(event, endpoint, eventNumber))
```

bzbarsky-apple, 2022-06-08:
This is not about "logging".  This is about Matter events.

ReneJosefsen, 2022-06-08:
This is an old section of code used for demonstration purposes. Since events are not going to make it for 1.0 I would think the easiest solution is to just remove the code that triggers the LogEvent. 

If that is acceptable, I can do a PR to fix this.

bzbarsky-apple, 2022-06-08:
In what sense are "events not going to make it for 1.0"?

ReneJosefsen, 2022-06-09:
I assumed that events in general was deferred for now, but there might be something I missed?

bzbarsky-apple, 2022-06-09:
Events in general are implemented and tested.

andy31415, 2022-06-09:
Fixed by 19379

closed 2022-06-09 by andy31415