project-chip/connectedhomeip #19306: [pre-sve] Chip-tool commissionable device discovery via BLE not working

pankore, 2022-06-08:
#### Problem
Platform: RPI4 chip-tool
SHA: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152
chip-tool is unable to discover commissionable devices via BLE
Required for TC-DD-3.1, TC-DD-3.19

Steps to reproduce:
- Commissionable device starts BLE adv (confirmed by nrfconnect ble app)
- `./chip-tool discover commissionables`

[discover-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8859868/discover-chip-tool.txt)

bzbarsky-apple, 2022-06-08:
See #18147 and #17963.

I just looked at TC-DD-3.1 and I don't see anything in there that requires `discover commissionables` to do anything with BLE.  Which step do you think needs it?

Similar for TC-DD-3.19: which step do you think needs `discover commissionables`?

mamehta, 2022-07-04:
The correct TC reference for this is TC-DD-2.1

bzbarsky-apple, 2022-07-11:
@mamehta Which step of TC-DD-2.1 needs anything like `discover commissionables`?  I am not seeing it.

mamehta, 2022-07-13:
In the test plan published at https://groups.csa-iot.org/wg/matter-csg/document/26538 (version 1.2) -
I'm seeing several steps in TC-DD-2.1 involving discover commissionables, like Step 2 below pasted as an example. I'm also seeing it in steps 11, 12, 14a, 15a and 15b.

Step 2:
1. Discover commissionables (example for over mDNS only)
./chip-tool discover commissionables

If (PICS_CHIP_DEV) TH should discover the DUT else it should not
If !(PICS_CHIP_DEV) instruct DUT to start Advertising and scan again for commissionale devices using TH
...

bzbarsky-apple, 2022-07-15:
@mamehta So the issue is with the specific verification steps for the test plan?  It's possible that those verification steps should be fixed.  The test plan itself does not involve listing discovered things; it just involves discovering something and connecting to it.

mamehta, 2022-07-15:
@bzbarsky-apple It's my understanding from reading the test case that `discover commissionables` is just a convenience to validate in step 2, etc. whether the DUT is discoverable without having to run the full commissioning flow test which would probably be more time consuming. If we don't intend for `discover commissionables` to be used for this test or if that command itself is not working properly, we could of course modify the test case to require the tester to run the full commissioning flow to verify that the DUT is getting discovered and commissioned.

Also, if I'm not mistaken, `discover commissionables` is intended for discovery over mDNS only, so perhaps the title of this issue is also misleading at this point.

bzbarsky-apple, 2022-07-18:
Right, `discover commissionables` is indeed intended for mDNS discovery only right now, so we should not be using it for BLE discovery.

mamehta, 2022-07-20:
The steps I'm pointing to, though, where it doesn't work are intended to validate mDNS discovery.

bzbarsky-apple, 2022-07-20:
OK.  So what is the actual issue here?  The issue summary is talking about BLE.  Is there an mDNS issue?

mamehta, 2022-07-20:
So, yes, to answer your question, this issue is actually about mDNS discovery, not BLE.

bzbarsky-apple, 2022-07-21:
> However, the said command doesn't work.

Please file an issue with details: exact steps executed and logs from both sides.

chrisbeach59, 2022-07-21:
I am seeing this same issue with my DUT BLE/Wifi.

cjandhyala, 2022-07-22:
@chrisbeach59 The 'chip-tool discover commissionables' is meant for mDNS discovery only right now. If you need to test the BLE , you will need to use HCIdump or 'nrfconnect app' or similar app. we will have the test plan author correct the issue. 

pls file a ticket in test plans repo if you don't see this fixed in the next version. since Its not an SDK issue, I am closing it. Pls re-open If you think its an SDK issue.

closed 2022-07-22 by cjandhyala

liamgonyea, 2022-07-22:
Verification steps for the applicable steps throughout the test plan have been updated to not use `./chip-tool discover commissionables` when discovering BLE device advertisements