project-chip/connectedhomeip #19322: Unable to clear fabric from credential - error noticed when trying to unpair M5 board

phonnakasturi-apple, 2022-06-08:
With SHA 5362125e31b8bdd7b2dddb2da381eb7bffe80f73

Unpair of M5 board throws -> "Unable to clear fabric from credential" error

**Steps to reproduce**

1. Install and launch chiptool app on iOS
2. Make sure M5 board is flashed
3. Pair M5 board with chipTool app by scanning QR code or by using short pin
4. Select Unpair Devices on chiptool app
5. Choose Unpair Devices

**Note:** Unpair still works but error is noticed

Attaching the complete logs from M5 board:
[unpairErrorLogs.txt](https://github.com/project-chip/connectedhomeip/files/8863377/unpairErrorLogs.txt)

bzbarsky-apple, 2022-06-08:
This looks like door lock bits.  @Morozov-5F could you take a look please?

bzbarsky-apple, 2022-06-08:
And in particular, it does not make sense to log errors in this loop:
```
    for (uint16_t userIndex = 1; userIndex <= maxNumberOfUsers; ++userIndex)
    {
        EmberAfPluginDoorLockUserInfo user;
        if (!emberAfPluginDoorLockGetUser(endpointId, userIndex, user))
        {
            ChipLogError(Zcl,
                         "[OnFabricRemoved] Unable to get the user - internal error [endpointId=%d,fabricIndex=%d,userIndex=%d]",
```
because not having all slots filled with users is _normal_, no?  But maybe the real issue here is just the bogus lock delegate in all-clusters-app?

Similar for this loop:
```
    for (uint16_t credentialIndex = startIndex; credentialIndex <= maxNumberOfCredentials; ++credentialIndex)
    {
        EmberAfPluginDoorLockCredentialInfo credential;
        if (!emberAfPluginDoorLockGetCredential(endpointId, credentialIndex, credentialType, credential))
        {
            ChipLogError(
```

Morozov-5F, 2022-06-09:
@bzbarsky-apple the door lock cluster expects that the `emberAfPluginDoorLockGetUser` and such return `false` only in case when the error occurred, when the user is missing it should be specified in the `userStatus` field of the `user`. This is [outlined](https://github.com/project-chip/connectedhomeip/blob/de9e9076303a11c568d2533b3e624a96db366dae/src/app/clusters/door-lock-server/door-lock-server.h#L787) in the function doxygen but in a twisted way, I admit.

@phonnakasturi-apple regarding the issue, if that was an `all-clusters-app` for esp32 then it is because those callbacks are not implemented and they return `false` by default.

bzbarsky-apple, 2022-06-09:
@Morozov-5F Perhaps the all-clusters-app callbacks should be changed to just return all users as not there instead of false?  And similarly for credentials?

Alternately, it could return 0 for max numbers of users/credentials supported?

Morozov-5F, 2022-06-09:
@bzbarsky-apple yeah, that's a great idea. Actually, I just realized that Linux all-clusters-app borrows the lock implementation from the Linux lock-app. Maybe we can make a minimal shared lock implementation for all-clusters-app?

bzbarsky-apple, 2022-06-10:
> Maybe we can make a minimal shared lock implementation for all-clusters-app that could be fit into more constrained platforms as well?

That would be pretty nice if we find the time, obviously, but for now we should just try to get to the point where all-clusters-app at least is not violating the spec, if we can do it easily.

PR #22102 (2022-09-08): [all-clusters-app][esp32] Add LockManager/LockEndpoint and the door lock plugin handlers methods to the example application
#### Problem

PR #18289 has moved the lock implementation out of the all-clusters-app-common build file. As a result the ESP32 version has a different behaviour than the linux version.

fix #19322 but that is only for the esp32 platform. In the long run and if we want to have a common implementation it would be nice to do an implementation that is dedicated to constrained platforms as suggested in https://github.com/project-chip/connectedhomeip/issues/19322#issuecomment-1 [... cut]
closed 2022-09-08 by andy31415