project-chip/connectedhomeip #19333: [pre-sve] [TC-SU-2.8][TC-Su-4.1] Transfer does not happen on TH2/OTA-P

Shadi-Silabs, 2022-06-08:
#### TC-SU-2.8 step#2
After killing the OTA Provider for Fabric 1, the transfer never happens on TH2/OTA-Provider.

 #### TC-SU-2.8 step#3
The transfer also never happens on TH2/OTA-Provider.

 #### TC-SU-4.1: 
This test is affected from the same problem of TC-SU-2.8.

 #### Please check the logs below:
[TC-SU-2.8 failed step#2.zip](https://github.com/project-chip/connectedhomeip/files/8864529/TC-SU-2.8.failed.step.2.zip)
[TC-SU-2.8 failed step#3.zip](https://github.com/project-chip/connectedhomeip/files/8864530/TC-SU-2.8.failed.step.3.zip)

franck-apple, 2022-06-09:
As agreed, bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

woody-apple, 2022-06-13:
Cert Blocker Review: Assigning to @carol-apple to follow up here to confirm this is a setup issue.

isiu-apple, 2022-06-15:
Verified the steps using Linux version.  Will upload successful logs shortly.  Steps taken:

Terminal 1:

Run OTA-P-1

out/apps/ota-provider/chip-ota-provider-app --discriminator 22 --secured-device-port 5565 --KVS /tmp/chip_kvs_provider -f ~/Downloads/test2.ota

Terminal 2:

Run OTA-P-2

out/apps/ota-provider/chip-ota-provider-app --discriminator 23 --secured-device-port 5566 --KVS /tmp/chip_kvs_provider2 -f ~/Downloads/test3.ota

Terminal 3:

Run OTA-R

out/apps/ota-requestor/chip-ota-requestor-app --discriminator 18 --secured-device-port 5560 --KVS /tmp/chip_kvs_requestor --periodicQueryTimeout 10

Terminal 4:

Commission OTA-P-1 into first fabric (alpha)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xC0FFEE 20202021 22

Commission OTA-P-2 into second fabric (beta)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xB0BA 20202021 23 --commissioner-name beta

Commission OTA-R into first fabric (alpha)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xDEB 20202021 18

Open Basic Commissioning Window for OTA-R

out/apps/chip-tool/chip-tool administratorcommissioning open-basic-commissioning-window 600 0xDEB 0 --timedInteractionTimeoutMs 600

Commission OTA-R into second fabric (beta)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xB0B 20202021 18 --commissioner-name beta

Write/Read DefaultOTAProviders on the first fabric (alpha)

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor write default-ota-providers '[{"providerNodeID": 12648430, "endpoint": 0}]' 0xDEB 0

 
 out/apps/chip-tool/chip-tool otasoftwareupdaterequestor read default-ota-providers 0xDEB 0

Write/Read DefaultOTAProviders on second fabric (beta)

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor write default-ota-providers '[{"providerNodeID": 45242, "endpoint": 0}]' 0xB0B 0 --commissioner-name beta

 out/apps/chip-tool/chip-tool otasoftwareupdaterequestor read default-ota-providers 0xB0B 0 --commissioner-name beta

Write ACL for OTA-P-1

out/apps/chip-tool/chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 0xC0FFEE 0

Write ACL for OTA-P-2

out/apps/chip-tool/chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 0xB0BA 0 --commissioner-name beta

Issue AnnounceOTAProvider command to OTA-R/OTA-P-1

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor announce-ota-provider 0xC0FFEE 0 0 0 0xDEB 0

Query Image should succeed.

Kill OTA-P-1

Issue another AnnounceOTAProvider command to OTA-R/OTA-P-1

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor announce-ota-provider 0xC0FFEE 0 0 0 0xDEB 0

Verify that the DUT sends a QueryImage command to TH2/OTA-P when failing to reach out to TH1/OTA-P after a finite number of unsuccessful attempts.

Issue AnnounceOTAProvider command to OTA-R/OTA-P-2

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor announce-ota-provider 0xB0BA 0 0 0 0xB0B 0

Query Image and image transfer from OTA-P-2 succeeds.

selissia, 2022-06-15:
Does the test plan have the correct steps? The document I have (Draft_TE9_TestPlanVerificationSteps 20220609.xlsx) does not list announce-ota-provider command in TC-SU-2.8 Steps 2 and 3.

sankarselvam, 2022-06-16:
@selissia We tried this tests and seems passing in QA. Logs attached for the reference. Can you please check once?

[OTA-REQUESTOR (1).txt](https://github.com/project-chip/connectedhomeip/files/8920986/OTA-REQUESTOR.1.txt)
[CHIP-TOOL (1).txt]
[OTA-PROVIDER-2.txt](https://github.com/project-chip/connectedhomeip/files/8920989/OTA-PROVIDER-2.txt)
(https://github.com/project-chip/connectedhomeip/files/8920987/CHIP-TOOL.1.txt)
[OTA-PROVIDER-1.txt](https://github.com/project-chip/connectedhomeip/files/8920990/OTA-PROVIDER-1.txt)

sankarselvam, 2022-06-16:
Attaching the commands list which have used for ease of Ref
[OTA-  MP CMMDS.txt](https://github.com/project-chip/connectedhomeip/files/8921142/OTA-.MP.CMMDS.txt)

isiu-apple, 2022-06-16:
Re-did the test on Linux without announce-ota-provider command and tests also passes.  Here are the steps taken:

Terminal 1:

Run OTA-P-1

out/apps/ota-provider/chip-ota-provider-app --discriminator 22 --secured-device-port 5565 --KVS /tmp/chip_kvs_provider -f ~/Downloads/test2.ota

Terminal 2:

Run OTA-P-2

out/apps/ota-provider/chip-ota-provider-app --discriminator 23 --secured-device-port 5566 --KVS /tmp/chip_kvs_provider2 -f ~/Downloads/test3.ota

Terminal 3:

Run OTA-R

out/apps/ota-requestor/chip-ota-requestor-app --discriminator 18 --secured-device-port 5560 --KVS /tmp/chip_kvs_requestor --periodicQueryTimeout 10

Terminal 4:

Commission OTA-P-1 into first fabric (alpha)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xC0FFEE 20202021 22

Commission OTA-P-2 into second fabric (beta)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xB0BA 20202021 23 --commissioner-name beta

Commission OTA-R into first fabric (alpha)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xDEB 20202021 18

Open Basic Commissioning Window for OTA-R

out/apps/chip-tool/chip-tool administratorcommissioning open-basic-commissioning-window 600 0xDEB 0 --timedInteractionTimeoutMs 600

Commission OTA-R into second fabric (beta)

out/apps/chip-tool/chip-tool pairing onnetwork-long 0xB0B 20202021 18 --commissioner-name beta

Write/Read DefaultOTAProviders on the first fabric (alpha)

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor write default-ota-providers '[{"providerNodeID": 12648430, "endpoint": 0}]' 0xDEB 0

 
 out/apps/chip-tool/chip-tool otasoftwareupdaterequestor read default-ota-providers 0xDEB 0

Write/Read DefaultOTAProviders on second fabric (beta)

out/apps/chip-tool/chip-tool otasoftwareupdaterequestor write default-ota-providers '[{"providerNodeID": 45242, "endpoint": 0}]' 0xB0B 0 --commissioner-name beta

 out/apps/chip-tool/chip-tool otasoftwareupdaterequestor read default-ota-providers 0xB0B 0 --commissioner-name beta

Write ACL for OTA-P-1

out/apps/chip-tool/chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 0xC0FFEE 0

Write ACL for OTA-P-2

out/apps/chip-tool/chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 0xB0BA 0 --commissioner-name beta

[Step 1] Wait for OTA-R (DUT) to send QueryImage command to OTA-P-1.  Image transfer succeeds.

Kill OTA-P-1

[Step 2] DUT sends a QueryImage command to TH1/OTA-P. TH1/OTA-P does not respond with QueryImageResponse.  Verify that the DUT sends a QueryImage command to TH2/OTA-P when failing to reach out to TH1/OTA-P after a finite number of unsuccessful attempts.

[Step 3] DUT sends a QueryImage command to TH2/OTA-P. TH2/OTA-P sends a QueryImageResponse back to the DUT. Status is set to UpdateAvailable. Verify there is a transfer of the software image from TH2/OTA-P to the DUT.

Note that in the Linux OTA-R implementation, OTA-R here chooses to retry OTA-P-1 first and fails after timeout and then moves on to try OTA-P-2 and succeeds.

franck-apple, 2022-06-17:
Per agreement with @isiu-apple and @sankarselvam, closing this issue.

closed 2022-06-17 by franck-apple

selissia, 2022-06-22:
@sankarselvam @isiu-apple : Still, this is a test specification issue. In both of your examples the periodic query timeout is set to a very low value (10 or 20 seconds) whereas by default it is 24 hrs. This is the reason the test fails with the default config. 
The test description must mention that the periodic query timeout must be set to a low value.