project-chip/connectedhomeip #19341: Video device types not up to date in `matter_devices.xml`

balloob, 2022-06-08:
#### Problem

I cross-referenced X with `matter_devices.xml` and found the following video related items to not be up to date:

- 10.2 Basic Video Player (0x0028) is missing from matter_devices.xml
- 10.3 Casting video player (0x0023) is called “Video Player” in matter_devices.xml
- 10.5 Content App is called “Content Application” in matter_devices.xml
- 10.5 Content Application in `matter_xml` incorrectly references cluster `TV Channel`. It should be `Channel`.
- 10.6 Casting Video Client is missing from matter_devices.xml
- 10.7 Video Remote Control is missing from matter_devices.xml

 #### Proposed Solution

Update `matter_devices.xml` according the latest spec.

lazarkov, 2022-06-09:
hi @balloob can you please specify where can I find the matter_devices.xml?

TV Channel was renamed 6 months ago. I don't know how you're still able to find this reference

balloob, 2022-06-09:
It's here:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/matter-devices.xml

_Tip: You can press T on any GitHub page to search the files in the repository._

naveenkommareddi, 2022-07-19:
We should label this issue with "Request Cert Blocker"

lazarkov, 2022-07-19:
@balloob
What should be the device id for these two?
10.6 Casting Video Client is missing from matter_devices.xml
10.7 Video Remote Control is missing from matter_devices.xml

balloob, 2022-07-20:
I do not know. I just noticed that it's missing.

closed 2022-08-03 by lazarkov