project-chip/connectedhomeip #19348: [pre-sve]Can not build Thermostat without scenes

chrisbeach59, 2022-06-08:
#### Problem
It is not possible to build an application that contains a thermostat cluster without the scenes server cluster being present
 * expected behavior
 * The scenes cluster is is optional for the thermostat device type.  It should be possible to build a minimal thermostat device type without the scenes cluster
 * actual behavior
 * Build fails
 * steps to reproduce
 * remove the scenes cluster from EP1 for the thermostat test app and attempt to build
 * system configuration
 * all
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution

andy31415, 2022-06-09:
@chrisbeach59 - could you share relevant portions of the logs of what error you get when build fails?

also please edit the template to remove the boilerplate (things like `all` and `screenshots, images, logs, etc.` seem to have no meaning for this bug report).

chrisbeach59, 2022-06-09:
Unable to reproduce with core sdk build now.   Closing

closed 2022-06-09 by chrisbeach59