project-chip/connectedhomeip #19351: Integer Overflow Leading to Buffer Overflow in `vFillBuffer()`

robszewczyk, 2022-06-08:
#### Problem
Code with a potential integer overflow leading to a buffer overflow:
````
static uint16_t vFillBuffer(uint8_t * buffer, uint16_t bufferLen, uint8_t frameControl, uint16_t manufacturerCode,
                                           CommandId commandId, const char * format, va_list argPointer)
{
    uint64_t value;
    uint8_t valueLen;
    uint8_t * data;
    uint16_t dataLen;
    uint8_t i;
    uint16_t bytes = 0;
    [...]
    if (bufferLen < bytes + dataLen + valueLen)
    {
        emberAfDebugPrintln("ERR: Buffer too short for %d bytes for format '%c'", dataLen +valueLen, cmd);
        return 0;
    [...]
    memcpy(buffer + bytes, data, dataLen);
````
As the types of bytes , dataLen and valueLen are all `uint16_t` or smaller, if they add up
to over `0xffff` this will cause an integer overflow and the `bufferLen` check could pass
when it should not. Later there’s a `memcpy()` call that could cause a buffer overflow
 #### Proposed Solution
Overflow should be checked with help of cast to a larger type. While casting
the leftmost summand variable will be enough due to the type promotion rule, it might be
less prone to future errors to just cast them all.

````
    if (bufferLen < (size_t)bytes + (size_t)dataLen + (size_t)valueLen)
    {
        emberAfDebugPrintln("ERR: Buffer too short for %d bytes for format '%c'", (uint32_t)dataLen + valueLen, cmd);
        return 0;
````

Note this is low impact as no untrusted parameters are being passed in as arguments.  Can be potentially fixed post 1.0

turon, 2022-07-15:
This appears to be in a third_party library: `third_party/silabs/gecko_sdk/protocol/zigbee/app/framework/util/client-api.c`.

Seeking SiLabs person to reassign.

bzbarsky-apple, 2022-07-15:
This code used to be in our tree, but was removed in #19291

closed 2022-07-15 by bzbarsky-apple