project-chip/connectedhomeip #19353: Potential for Command Injection in Linux Connectivity Manager

robszewczyk, 2022-06-08:
#### Problem
It is very unlikely for the described vulnerability to have any impact, because it would
require unusual configurations or attacker to have high privileges.

Similar code patterns can often be trivially exploited to achieve command execution in
context of target process.

The following excerpt of code that takes a network interface name and uses it as a part of
a shell command:
````
#define CHIP_DEVICE_CONFIG_LINUX_DHCPC_CMD "dhclient -nw %s"
[...]
sprintf(cmdBuffer, CHIP_DEVICE_CONFIG_LINUX_DHCPC_CMD, sWiFiIfName);
int dhclientSystemRet = system(cmdBuffer);
````

While the intention is that the interface name is an argument to `dhclient` there are ways of
injecting additional commands due to shell giving special meaning to many non-alphanumeric characters.

Please note that even inputs like network interface name could be
avenues for command injection in some circumstances. Interface names can contain a
variety of characters and could be created with SetUID tools present on the system.

Following commands show it is possible to create a network interface name containing
many characters that are special to shells and that there might be circumstances where
network interfaces could be created by non-root users.

````
$ docker network create --opt com.docker.network.bridge.name='foo;${x[y(*z' testname4
$ ls -l /sys/class/net/
[...]
lrwxrwxrwx 1 root root 0 May 17 12:59 'foo;${x[y(*z' -> '../../devices/virtual/net/foo;$
{x[y(*z'
````
 #### Proposed Solution
Recommend against usage of `system()` and `popen()` which are prone to
shell command injection in multiple ways.

In general, when an external command needs to be executed, recommend using
one of `exec()` calls. When using those one should still be aware of argument injection, for
example `-xfoo` might inject an option `x` with argument `foo` .

robszewczyk, 2022-06-08:
Another item is perhaps best addressed post 1.0

Alami-Amine, 2025-01-23:
These are Security Audit findings, they should be at least addressed or examined