project-chip/connectedhomeip #19359: ExchangeContext may leak in `ExchangeContext::OnSessionReleased`

kghost, 2022-06-09:
#### Problem
When `OnSessionReleased` is called while the exchange is in state of WillSendMessage, the `OnExchangeClosing` is triggered, but the ref count is not decreased. Will cause that exchange leak.

 #### Proposed Solution
To prevent possible problem like this one, and make exchange context more robust
* Implement a state machine to manage ref counters
* Make Close/Abort re-entrant

bzbarsky-apple, 2022-06-09:
So actually, wait.  In the case when the exchange is waiting for a message send and the session expires, what are the intended refcount semantics?  Right now it will call OnExchangeClosing, but that does not do any releasing of the exchange.  What will release it is when the consumer tries to send the message, the message send _fails_ and the consumer then closes the exchange.  So there is no leak in that case either.

I do think the logic here could be clearer and better documented.  I'll do a PR to address that.

bzbarsky-apple, 2022-06-09:
And in particular: if the exchange is waiting for a response, the refcount is held by the network stack.  The exchange can release that refcount itself.  If the exchange is waiting for a send, the refcount is held by the consumer that will do the send.  The exchange has no way to release that refcount itself without leaving dangling pointers.

kghost, 2022-06-09:
> If mResponseTimeout is 0, that means "stay open indefinitely waiting for a response" and the responsibility for closing the exchange is then on the consumer.

> if the exchange is waiting for a response, the refcount is held by the network stack.

These 2 statement just conflict.

bzbarsky-apple, 2022-06-09:
OK, good point.  If the exchange is waiting for a response and a response timeout was requested, the network stack will handle closing the exchange on response timeout, unless the consumer takes action to prevent it.  The consumer will be notified when the timeout happens, so it can drop its ref to the exchange.

If the exchange is waiting on a response with no timeout allowed, the consumer needs to handle closing the exchange and dropping the reference to it, generally.  In the "session released" case we treat things as a timeout, looks like.

I agree this is all a bit more complicated than one would prefer.

closed 2023-06-26 by bzbarsky-apple