project-chip/connectedhomeip #19376: Client test-cases - The attribute value is mismatching

sowmyassp, 2022-06-09:
**Issue** : Attribute-list value is mismatched when we read the attribute-list attribute 

**Expected outcome** : Should list the mandatory attribute of respective cluster 

**Actual behaviour** : It list only attribute which as conformance in spec as M of respective cluster

**For example** : 
On-Off cluster - The attribute-list is read using all-clusters-app is showing 10 entries including all attributes, where the attribute-list is read using all-clusters-minimal-app which is used for turning off the optional attribute shows 6 entries i.e., attribute (GlobalSceneControl, OnTime, OffWaitTime, StartUpOnOff) is not listed when we read attribute list using all-clusters-minimal-app which is mandatory 

**Clusters affected due to this** : 
1. On/Off Cluster - TC-OO-3.1
2. Level Control Cluster - TC-LVL-2.3
3. Switch Cluster - TC-SWTCH-3.1
4. Mode Select Cluster - TC-MOD-1.3

**CommitID** : 2d1f8a1eff8602f1f4ee85114d3d86accb25117d	

**Test platform** :
1. Chip-tool - MacOS
2. DUT - RPI - RPI-4, 8GB RAM	
3. all-cluster-app/all-clusters-minimal-app - linux platform	

PFA Logs : 
[clusters_minimal_app.txt](https://github.com/project-chip/connectedhomeip/files/8870306/clusters_minimal_app.txt)
[all_clusters_app.txt](https://github.com/project-chip/connectedhomeip/files/8870308/all_clusters_app.txt)

bzbarsky-apple, 2022-06-09:
> attribute (GlobalSceneControl, OnTime, OffWaitTime, StartUpOnOff) is not listed when we read attribute list using all-clusters-minimal-app which is mandatory

These attributes are not mandatory.  They're only allowed if the LT feature is enabled.  The LT feature is not enabled in all-clusters-minimal-app, because it's a _minimal_ app that disables everything that can be disabled.

> Actual behaviour : It list only attribute which as conformance in spec as M of respective cluster

Yes, that is the intent of the minimal app.  It only includes things that are mandatory.

bzbarsky-apple, 2022-06-09:
@raju-apple @cjandhyala @sowmyassp @dwgaul please do let me know if I'm missing something about the request in https://github.com/project-chip/connectedhomeip/issues/16851, and reopen the issue if so.