project-chip/connectedhomeip #19378: Chip-tool failure reading measured temperature.

AntonGrey, 2022-06-09:
#### Chiptool get Response Failure at reading some Temperature Measurement cluster's attributes, such as MesuredValue, MinMesuredValue, MaxMesuredValue.

 * expected behavior: Success reading temperature attributes.
 * actual behavior: Chip-tool failures at response processing with `Response Failure: Can not decode Data`
 * steps to reproduce: read any non-null Temperature attribute with a command like `./chip-tool temperaturemeasurement read measured-value 1 1`

 #### Proposed Solution
Looks like chip-tool expects values in degrees Celsius, but actually Measured Value calculated as `MeasuredValue = 100 x temperature [°C]`

Chip-tool log:
[chiptool-failure-read-temperature.log](https://github.com/project-chip/connectedhomeip/files/8870603/chiptool-failure-read-temperature.log)

bzbarsky-apple, 2022-06-09:
@AntonGrey I can't reproduce this issue on SHA 3fafeceff1681d01244c602584da47bd6f3bfd90 using chip-all-clusters-app as the server.

What server app are you using, and what SHA are you using for both the server app and chip-tool?

> Looks like chip-tool expects values in degrees Celsius

chip-tool does not attach any semantics to values.  

Looking at the log, the response that comes back has "Data = 2000" as some numeric type.  When chip-tool is failing, at `src/lib/core/CHIPTLVReader.cpp:185`, it's trying to read the data as a signed integer, but the type is not a signed integer.  My guess would be that the server encoded as an unsigned integer.

The type for MeasuredValue in the spec is `int16`, so reading it as a signed integer is correct.  The question is why the server encoded something that is not a signed integer.

AntonGrey, 2022-06-09:
@bzbarsky-apple Thanks for explanation. I was using not an example, just our Bridge implementation. 
Really, there is no issue in chip-tool. 
The issue was on server side, MeasuredValue was encoded as unsigned integer.