project-chip/connectedhomeip #19387: Canceling timers does not work reliably in the SystemLayerSelect implementation

bzbarsky-apple, 2022-06-09:
#### Problem
When we process timers, we grab a batch out of the list before we start calling them, and now the timers in that batch cannot be canceled.

 #### Proposed Solution
Either stop doing the batch thing and process timers one by one (maybe with a cap on how far down the list we go), or make the batch a member so that canceling can remove from the batch too.

bzbarsky-apple, 2022-06-10:
Once this is fixed, the workaround for this in `FailSafeContext::FailSafeTimerExpired` can be removed, I think.

kpschoedel, 2022-06-10:
> I think the code can be identical to that in `FreeRTOS::HandlePlatformTimer()`

It can't, because in the Select implementation a timer callback that re-adds itself with immediate effect (e.g. using `System::Layer::ScheduleWork()`, not to be confused with `PlatformMgr::ScheduleWork()`) will steal all the execution slots. Probably have to use the cancel-from-batch method.

bzbarsky-apple, 2022-06-10:
Ah, and the FreeRTOS setup does not have that problem?

kpschoedel, 2022-06-10:
> Ah, and the FreeRTOS setup does not have that problem?

On FreeRTOS, `ScheduleWork` uses the recent `ScheduleLambda<>()`, which percolates through to `PlatformMgr().PostEvent()`, rather than the timer queue. In the Select version, it's approximately `StartTimer(0, …)`.

I don't know why the Select version doesn't use `ScheduleLambda`, but I seem to remember that originally only existed on LwIP platforms, so that might be the only reason. @kghost

kpschoedel, 2022-06-10:
It turns out there's another subtlety with the Select implementation and being able to Cancel from the expired timers.  Suppose timers A and B expire, and A's callback wants to schedule B. This cancels B from the expired timer list and re-adds it to the main list, so it doesn't execute in the current batch. This can happen forever.

For `StartTimer()`, the cancel/re-add behaviour is documented in the API, but for `ScheduleWork()` it isn't, and this actually occurs in a unit test (`TestReadHandler::KillOverQuotaSubscriptions`). Treating `ScheduleWork()` as `StartTimer(0)` is probably just wrong.

closed 2022-09-05 by andy31415