project-chip/connectedhomeip #19409: M5Stack didn't receive messages over WiFi.

kpark-apple, 2022-06-09:
#### Problem

In the attached pcap log, M5stack didn't respond to the following messages (sigma1 message and retransmission). Note that those destination addresses were discovered in the commissioner.

```
1245	17:56:19.564337	fd42:31f:824c:45bd:1485:c3a0:7d53:e9b6	fd42:31f:824c:45bd:baf0:9ff:fec4:8274	UDP	239	56438 → 5540 Len=177[Packet size limited during capture]	0.037289	68.527781
1402	17:56:24.884556	fd42:31f:824c:45bd:1485:c3a0:7d53:e9b6	fd42:31f:824c:45bd:baf0:9ff:fec4:8274	UDP	239	56438 → 5540 Len=177[Packet size limited during capture]	0.028272	73.848000
```

Also note that in the same pcap log, there were no responses to neighbor solicitation messages either for the same address:

```
1409	17:56:25.251689	fd42:31f:824c:45bd:1485:c3a0:7d53:e9b6	fd42:31f:824c:45bd:baf0:9ff:fec4:8274	ICMPv6	86	Neighbor Solicitation for fd42:31f:824c:45bd:baf0:9ff:fec4:8274 from f6:e2:e2:77:0d:74	0.197129	74.215133
1423	17:56:26.262943	fd42:31f:824c:45bd:1485:c3a0:7d53:e9b6	fd42:31f:824c:45bd:baf0:9ff:fec4:8274	ICMPv6	86	Neighbor Solicitation for fd42:31f:824c:45bd:baf0:9ff:fec4:8274 from f6:e2:e2:77:0d:74	0.069838	75.226387
```

While later M5Stack responded to link local address of the same IID:

```
1939	17:57:10.062727	fe80::4be:7e8b:3a4e:d0bc	fe80::baf0:9ff:fec4:8274	UDP	239	56438 → 5540 Len=177[Packet size limited during capture]	0.000159	119.026171
1962	17:57:11.674178	fe80::baf0:9ff:fec4:8274	fe80::4be:7e8b:3a4e:d0bc	UDP	567	5540 → 56438 Len=505[Packet size limited during capture]	0.267286	120.637622
```

M5stack seemed to have an issue receiving for `fd42:31f:824c:45bd:baf0:9ff:fec4:8274` address only.

[packets-06-08-2022-175854-911883543-0.pcapng.zip](https://github.com/project-chip/connectedhomeip/files/8873560/packets-06-08-2022-175854-911883543-0.pcapng.zip)
[M5-pairing-fail.txt](https://github.com/project-chip/connectedhomeip/files/8873562/M5-pairing-fail.txt)

bzbarsky-apple, 2022-06-10:
@dhrishi

dhrishi, 2022-06-11:
Thanks! We will take a look
cc @shubhamdp

closed 2022-12-30 by stale[bot]