project-chip/connectedhomeip #19414: [Group] [ACL] Secure Channel Test seems broken

jepenven-silabs, 2022-06-09:
#### Problem
TC-SC-5.1 is failing on multiples platform ( at least EFR32 and Linux all-cluster-app).

After the ACL write, device is unable to respond to all commands. Including unicast commands.

 #### Proposed Solution
Investigate it to see why this test is failing and to narrow down if it's a test plan issue (wrong command) or a corner case in the ACL code.

[tc-sc-5.1 step 1 acl failure.txt](https://github.com/project-chip/connectedhomeip/files/8874111/tc-sc-5.1.step.1.acl.failure.txt)

bzbarsky-apple, 2022-06-10:
So the relevant command is this:
```
pi@MatterRPi5:~$ ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{"cluster": null, "endpoint": 0, "deviceType": null}]},            {"fabricIndex": 1, "privilege": 3, "authMode": 3, "subjects": null, "targets": [{"cluster": null, "endpoint": 1, "deviceType": null}]}]' 1 0
```
That command defines the following ACLS:
1. Administer privilege over CASE for endpoint 0 only.
2. Operate privilege with group auth mode for any group for endpoint 1 only.

Then the test tries to issue a "toggle" command over CASE to endpoint 1.  That of course should fail the ACL check given the ACL above.

So the behavior of the SDK seems correct...  I don't know where the actual steps here came from.

Looking at the test plan at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/group_communication.adoc#131-tc-sc-51-adding-member-to-a-group---th-as-admin-and-dut-as-group-member it says to set some ACLs first (but not what they should be), then to do some operations on endpoint 0.  Those operations on endpoint 0 should work with the ACLs from the above command.

@jepenven-silabs

jepenven-silabs, 2022-06-10:
@bzbarsky-apple Seems like the issue is more test plan related in that case. Moving issues to the test-plan repo and closing this one for now.

https://github.com/CHIP-Specifications/chip-test-plans/issues/1452
https://github.com/CHIP-Specifications/chip-test-plans/issues/1453

closed 2022-06-10 by jepenven-silabs