project-chip/connectedhomeip #19419: Pre SVE Testing: TC-I-3.2 step 4 giving 0x85 error

Garik098, 2022-06-09:
As per step 4 of TC-I-3.2, DUT sends a TriggerEffect command to the Test Harness, with any supported EffectIdentifier argument and EffectVariant set to 0 .

Sent the command - ./chip-tool identify trigger-effect 0 0 1 1 

Response shows 0x85 error:
[1654816055916] [2257:2570756] CHIP: [DMG] ICR moving to [ResponseRe]
[1654816055916] [2257:2570756] CHIP: [DMG] InvokeResponseMessage =
[1654816055916] [2257:2570756] CHIP: [DMG] {
[1654816055916] [2257:2570756] CHIP: [DMG] 	suppressResponse = false, 
[1654816055916] [2257:2570756] CHIP: [DMG] 	InvokeResponseIBs =
[1654816055916] [2257:2570756] CHIP: [DMG] 	[
[1654816055916] [2257:2570756] CHIP: [DMG] 		InvokeResponseIB =
[1654816055916] [2257:2570756] CHIP: [DMG] 		{
[1654816055916] [2257:2570756] CHIP: [DMG] 			CommandStatusIB =
[1654816055916] [2257:2570756] CHIP: [DMG] 			{
[1654816055916] [2257:2570756] CHIP: [DMG] 				CommandPathIB =
[1654816055916] [2257:2570756] CHIP: [DMG] 				{
[1654816055916] [2257:2570756] CHIP: [DMG] 					EndpointId = 0x1,
[1654816055916] [2257:2570756] CHIP: [DMG] 					ClusterId = 0x3,
[1654816055916] [2257:2570756] CHIP: [DMG] 					CommandId = 0x40,
[1654816055916] [2257:2570756] CHIP: [DMG] 				},
[1654816055916] [2257:2570756] CHIP: [DMG] 				
[1654816055916] [2257:2570756] CHIP: [DMG] 				StatusIB =
[1654816055916] [2257:2570756] CHIP: [DMG] 				{
[1654816055916] [2257:2570756] CHIP: [DMG] 					status = 0x85 (INVALID_COMMAND),
[1654816055916] [2257:2570756] CHIP: [DMG] 				},
[1654816055916] [2257:2570756] CHIP: [DMG] 				
[1654816055916] [2257:2570756] CHIP: [DMG] 			},
[1654816055916] [2257:2570756] CHIP: [DMG] 			
[1654816055916] [2257:2570756] CHIP: [DMG] 		},
[1654816055916] [2257:2570756] CHIP: [DMG] 		
[1654816055916] [2257:2570756] CHIP: [DMG] 	],
[1654816055916] [2257:2570756] CHIP: [DMG] 	
[1654816055916] [2257:2570756] CHIP: [DMG] 	InteractionModelRevision = 1
[1654816055916] [2257:2570756] CHIP: [DMG] },
[1654816055916] [2257:2570756] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_0003 Command=0x0000_0040 Status=0x85
[1654816055916] [2257:2570756] CHIP: [TOO] Error: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)
[1654816055916] [2257:2570756] CHIP: [DMG] ICR moving to [AwaitingDe]
[1654816055916] [2257:2570756] CHIP: [EM] Sending Standalone Ack for MessageCounter:46388802 on exchange 4805i
[1654816055916] [2257:2570756] CHIP: [IN] Prepared secure message 0x7000075064f0 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 4805i with MessageCounter:40490913.
[1654816055916] [2257:2570756] CHIP: [IN] Sending encrypted msg 0x7000075064f0 with MessageCounter:40490913 to 0x0000000000000001 (1) at monotonic time: 000000000EE7B721 msec
[1654816055916] [2257:2570756] CHIP: [EM] Flushed pending ack for MessageCounter:46388802 on exchange 4805i
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the commissioner
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the controller
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the commissioner
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the controller
[1654816055916] [2257:2570751] CHIP: [IN] Expiring all connections for fabric 1!!
[1654816055916] [2257:2570751] CHIP: [IN] SecureSession[0x7f8561810280]: MarkForRemoval Type:2 LSID:22200
[1654816055916] [2257:2570751] CHIP: [IN] SecureSession[0x7f8561810280]: Released - Type:2 LSID:22200
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the commissioner
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the controller
[1654816055916] [2257:2570751] CHIP: [IN] Expiring all connections for fabric 2!!
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the commissioner
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the controller
[1654816055916] [2257:2570751] CHIP: [IN] Expiring all connections for fabric 3!!
[1654816055916] [2257:2570751] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1654816055916] [2257:2570751] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654816055916] [2257:2570751] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654816055916] [2257:2570751] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654816055916] [2257:2570751] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654816055916] [2257:2570751] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1654816055916] [2257:2570751] CHIP: [BLE] CancelConnection
[1654816055916] [2257:2570751] CHIP: [DL] Inet Layer shutdown
[1654816055916] [2257:2570751] CHIP: [DL] BLE shutdown
[1654816055916] [2257:2570751] CHIP: [DL] System Layer shutdown
[1654816055916] [2257:2570751] CHIP: [TOO] Run command failure: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)

bzbarsky-apple, 2022-06-10:
@Garik098 What SHA is being used?  What server application is being used?

I just tried this with ac8bdbf23b96216ed12a1f3a2981f6ee3647fffa and all-clusters-app as the server, and the command succeeds.

bzbarsky-apple, 2022-06-10:
From reporter: this was tested on 5d8599d195a93dfe5473f64e4f888c322e5df1d2 with all-clusters-app on an nrfconnect board.

`git grep Identify examples/all-clusters-app | grep nrfconnect` comes up empty, unlike `git grep Identify examples/all-clusters-app | grep linux`; the latter shows an `Identify` instance being set up and callbacks for it and whatnot.

Sounds like the nrfconnect version needs to add that.

@kkasperczyk-no could you take a look?

closed 2022-06-13 by andy31415