project-chip/connectedhomeip #19421: Pre SVE Testing: TC-OO-3.1 Step 3 - Writing 64, 65, 66 values on startuponoff attribute is giving the constraint error

Garik098, 2022-06-09:
As per step 3 of TC-OO-3.1, writing values 64 or 65 or 66 on startuponoff attribute is giving constraint error on chip tool. 

[image]

The response for the first three (0, 1, 2) is successful but for the last three (0x40, 0x41 and 0x42) is giving Constraint error (0x87)

Command used: ./chip-tool onoff write start-up-on-off 64 1 1

Response: 

[1654818714508] [3078:2612694] CHIP: [DMG] WriteClient moving to [ResponseRe]
[1654818714508] [3078:2612694] CHIP: [DMG] WriteResponseMessage =
[1654818714508] [3078:2612694] CHIP: [DMG] {
[1654818714508] [3078:2612694] CHIP: [DMG] 	AttributeStatusIBs =
[1654818714508] [3078:2612694] CHIP: [DMG] 	[
[1654818714508] [3078:2612694] CHIP: [DMG] 		AttributeStatusIB =
[1654818714508] [3078:2612694] CHIP: [DMG] 		{
[1654818714508] [3078:2612694] CHIP: [DMG] 			AttributePathIB =
[1654818714508] [3078:2612694] CHIP: [DMG] 			{
[1654818714508] [3078:2612694] CHIP: [DMG] 				Endpoint = 0x1,
[1654818714508] [3078:2612694] CHIP: [DMG] 				Cluster = 0x6,
[1654818714508] [3078:2612694] CHIP: [DMG] 				Attribute = 0x0000_4003,
[1654818714508] [3078:2612694] CHIP: [DMG] 			}
[1654818714508] [3078:2612694] CHIP: [DMG] 				
[1654818714508] [3078:2612694] CHIP: [DMG] 			StatusIB =
[1654818714508] [3078:2612694] CHIP: [DMG] 			{
[1654818714508] [3078:2612694] CHIP: [DMG] 				status = 0x87 (CONSTRAINT_ERROR),
[1654818714508] [3078:2612694] CHIP: [DMG] 			},
[1654818714508] [3078:2612694] CHIP: [DMG] 			
[1654818714508] [3078:2612694] CHIP: [DMG] 		},
[1654818714508] [3078:2612694] CHIP: [DMG] 		
[1654818714508] [3078:2612694] CHIP: [DMG] 	],
[1654818714508] [3078:2612694] CHIP: [DMG] 	
[1654818714508] [3078:2612694] CHIP: [DMG] 	InteractionModelRevision = 1
[1654818714508] [3078:2612694] CHIP: [DMG] }
[1654818714508] [3078:2612694] CHIP: [DMG] WriteClient moving to [AwaitingDe]
[1654818714509] [3078:2612694] CHIP: [TOO] Response Failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)
[1654818714509] [3078:2612694] CHIP: [EM] Sending Standalone Ack for MessageCounter:117138516 on exchange 52514i
[1654818714509] [3078:2612694] CHIP: [IN] Prepared secure message 0x700001cd64f0 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 52514i with MessageCounter:213613675.
[1654818714509] [3078:2612694] CHIP: [IN] Sending encrypted msg 0x700001cd64f0 with MessageCounter:213613675 to 0x0000000000000001 (1) at monotonic time: 000000000F104807 msec
[1654818714509] [3078:2612694] CHIP: [EM] Flushed pending ack for MessageCounter:117138516 on exchange 52514i
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the commissioner
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the controller
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the commissioner
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the controller
[1654818714509] [3078:2612688] CHIP: [IN] Expiring all connections for fabric 1!!
[1654818714509] [3078:2612688] CHIP: [IN] SecureSession[0x7fd03203aa80]: MarkForRemoval Type:2 LSID:3850
[1654818714509] [3078:2612688] CHIP: [IN] SecureSession[0x7fd03203aa80]: Released - Type:2 LSID:3850
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the commissioner
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the controller
[1654818714509] [3078:2612688] CHIP: [IN] Expiring all connections for fabric 2!!
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the commissioner
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the controller
[1654818714509] [3078:2612688] CHIP: [IN] Expiring all connections for fabric 3!!
[1654818714509] [3078:2612688] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1654818714509] [3078:2612688] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654818714509] [3078:2612688] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654818714509] [3078:2612688] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654818714509] [3078:2612688] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654818714509] [3078:2612688] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1654818714509] [3078:2612688] CHIP: [BLE] CancelConnection
[1654818714509] [3078:2612688] CHIP: [DL] Inet Layer shutdown
[1654818714509] [3078:2612688] CHIP: [DL] BLE shutdown
[1654818714509] [3078:2612688] CHIP: [DL] System Layer shutdown
[1654818714509] [3078:2612688] CHIP: [TOO] Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

bzbarsky-apple, 2022-06-10:
The spec for StartUpOnOff says:

> If the value is null, the OnOff attribute is set to its previous value. Otherwise, the behavior is defined in the table defining StartUpOnOffEnum.

That table, section "1.5.5.1. StartUpOnOffEnum" of the appclusters spec, defines the valid values of the enum to be 0, 1, 2.

Why are you trying to write 64, 65, 66 to this attribute, and why were you expecting a response different from CONSTRAINT_ERROR?  @Garik098

Garik098, 2022-06-10:
I agree with the statement. Sorry got confused between onoff value and startuponoff values. Looks like onoff supports the 64,65,66 while startuponoff only supports 0,1,2. May be we should change the command in the verification steps document as it asks to check in the latter values. Closing the ticket.

closed 2022-06-10 by Garik098

bzbarsky-apple, 2022-06-10:
OnOff is readonly, so writing to OnOff should just fail no matter what....