project-chip/connectedhomeip #19427: [darwin-framework-tool] timed Interactions only work with commands that require them

dwgaul, 2022-06-10:
Noticed that while testing with darwin-framework-tool that timed interactions (adding --timedInteractionTimeoutMs 200) don't appear to be working with all commands. For example:

```
 ./darwin-framework-tool onoff off 0x00000000000008B9 1 --timedInteractionTimeoutMs 200 
```
Will properly send the off command to the server, but on the server side, I do not see any `TimedRequestMessage` log lines.

However running:

```
./darwin-framework-tool doorlock lock-door 0x00000000000008B9 1 --timedInteractionTimeoutMs 200
```

Results in the actual timed request being sent to the server.

Not sure if this is really a big deal, but the standard Chip-tool sends a timedInteractionRequest for any command if the `--timedInteractionTimeoutMs` argument is used, regardless of the command.

bzbarsky-apple, 2022-06-10:
This is not intended behavior.  It's a bug in the fix for #17998.  Fix coming up.

bzbarsky-apple, 2022-06-10:
And for what it's worth, the argument was ignored for commands that don't take arguments, regardless of whether they require a timed invoke or not.

closed 2022-06-10 by bzbarsky-apple