project-chip/connectedhomeip #19444: "abort() was called at PC 0x400df2e9 on core 0" on M5 while adding it to iOS home app

shah-kartik, 2022-06-10:
abort() was called at PC 0x400df2e9 on core 0
0x400df2e9: chip::SessionHolder::Get() const at /Users/apple/connectedhomeip/src/lib/support/CodeUtils.h:489
 (inlined by) chip::Optional<chip::ReferenceCountedHandle<chip::Transport::Session> >::Value() const & at /Users/apple/connectedhomeip/src/lib/core/Optional.h:171
 (inlined by) chip::SessionHolder::Get() const at /Users/apple/connectedhomeip/src/transport/SessionHolder.h:54

Backtrace:0x40081e26:0x3ffe85400x400923a9:0x3ffe8560 0x40098ac6:0x3ffe8580 0x400df2e9:0x3ffe85f0 0x4012da31:0x3ffe8610 0x400dbfb3:0x3ffe8640 0x400dc545:0x3ffe8660 0x400dc619:0x3ffe8790 0x400dc675:0x3ffe87d0 0x40137e41:0x3ffe87f0 0x40137e51:0x3ffe8810 0x4013b216:0x3ffe8830 0x4013b4ee:0x3ffe8850 0x4013b515:0x3ffe88c0 0x40095631:0x3ffe88e0 
0x40081e26: panic_abort at /Users/apple/tools/esp-idf/components/esp_system/panic.c:402

0x400923a9: esp_system_abort at /Users/apple/tools/esp-idf/components/esp_system/esp_system.c:121

0x40098ac6: abort at /Users/apple/tools/esp-idf/components/newlib/abort.c:46

the full log is attached her
[M5_Panic.log](https://github.com/project-chip/connectedhomeip/files/8876609/M5_Panic.log)
e

bzbarsky-apple, 2022-06-10:
@shah-kartik What SHA was used here?  Knowing that is pretty important in terms of diagnosing the problem...  Clearly we are trying to report a ReadHandler whose underlying secure session has gone away, but the reasons secure sessions can go away have changed over time, so this may or may not still be an issue on tip, and the fix needed may be different depending on the situation.

bzbarsky-apple, 2022-06-10:
Duplicate of #19472

closed 2022-06-10 by bzbarsky-apple