project-chip/connectedhomeip #19445: Matter Specification

nahidakbar, 2022-06-10:
I am sold on the promise and am very very very interested in using this for a few personal projects.

How does one get their hands on the matter specification? There is a lot of marketing saying its all open but I can't find any downloadable document. Do I have to read code?

bzbarsky-apple, 2022-06-10:
@clapre could you please point @nahidakbar to the right process for getting the spec?

clapre, 2022-06-10:
@nahidakbar specifications that are in draft status are only shared with Participant-level members. It is final versions of specifications that are shared more widely.
Please reach out to me if you wish to join. Thanks.

nahidakbar, 2022-06-10:
@clapre @bzbarsky-apple thanks for response.

I wasn't planning to spend 20k usd to become a participant level member. It's a bit beyond my weekend tinkering budget.

Is there an ETA on when the specs will be public?

clapre, 2022-06-11:
@nahidakbar I would think that as a tinkerer the open-source reference implementation would be more useful, and the specs are really for making compliant products?
The specification is on track to be finalized in the fall.

kaiyazeera, 2022-08-12:
Hi there, I would also like to have access because I am writing a thesis on Matter.
Having the code is nice, but I would really need access to the specifications for some of the work. I think even for developers there are many use cases where you would want to have access to the specification instead of the code.

I already sent an e-mail to CSA in which I kindly asked for access and got no response until now.
Can you provide information whether it is possible to get access for research purposes? It would be possible for me to sign a non-disclosure agreement.

kaiyazeera, 2022-08-12:
And, if this would not be possible, can you please explain why, and would you know a bit more precise which time in fall the specification will be released, and if it is released, will it be fully released or only partially?

clapre, 2022-08-26:
@kaiyazeera I have responded to your email now. Thanks for your interest.

closed 2022-08-26 by clapre

1technophile, 2022-10-04:
Specifications can be requested below:
https://csa-iot.org/developer-resource/specifications-download-request/

kaiyazeera, 2022-10-05:
Thank you very much for the notification :)