project-chip/connectedhomeip #19446: [pre-sve] TC-SC-4.3 TH is doesnot start advertising a service using DNS-SD

PSONALl, 2022-06-10:
#### Problem
In test case SC-4.3 step 3:

 * expected behavior
         - TH is instructed to start advertising a service using DNS-SD and using only the TXT records defined in section of the matter spec
 * actual behavior
         - TH doesn't start advertising service over dns-sd
 
 * system configuration
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * logs 
       - th logs: [th.txt](https://github.com/project-chip/connectedhomeip/files/8877852/th.txt)
       - dns-query logs: [dns-qwery.txt](https://github.com/project-chip/connectedhomeip/files/8877850/dns-qwery.txt)
       - esp32 logs: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8877837/esp32.txt)

bzbarsky-apple, 2022-06-10:
We can certainly add `enableServerInteractions` to chip-tool so it advertises...  But I would guess that in practice the TH here would not be chip-tool but rather one of the test apps, right?

That said, this specific test plan makes no sense to me.  I filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1454 on that.

franck-apple, 2022-06-10:
Bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

sumaky, 2022-06-14:
We can close this issue as this is tracked in test plan issue #1454.

franck-apple, 2022-06-14:
Closing per @sumaky 's comment.

closed 2022-06-14 by franck-apple