project-chip/connectedhomeip #19447: [TC-WNCV-2.4]-TH fails to read Type attribute from DUT based on FeatureMap value.

Rajashreekalmane, 2022-06-10:
#### Problem
As per Test-Plan, if PA_LF and PA_TL Feature is supported in Window Covering Cluster, if TH try to read Type attribute 
from DUT based on FeatureMap value currently TH reads 0 instead of set to 6 or 8 while the device supports  PA_LF and PA_TL.

 * **expected behavior:-**TH should read type atrribute value as 6 or 8
 * **actual behavior:-** TH reading type atrribute value as 0
 * **steps to reproduce:-**
 ./chip-tool windowcovering read feature-map 1 1
 ./chip-tool windowcovering read type 1 1
 
**system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:0ef4a9d97f70cb0f409d6736ec9c79217b691153
Dut application: all-clusters-app

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/window_covering.adoc#tc-wncv-2-4-type-attribute-dut_server

**Spec reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/WindowCovering.adoc#features

PFA log below:-
[featuremap.txt](https://github.com/project-chip/connectedhomeip/files/8877924/featuremap.txt)

bzbarsky-apple, 2022-06-10:
Thank you for including the commit id!

This was fixed in https://github.com/project-chip/connectedhomeip/pull/19311, which post-dates that commit id.

bzbarsky-apple, 2022-06-10:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/19163

closed 2022-06-10 by bzbarsky-apple