project-chip/connectedhomeip #19448: [pre-sve] Write on label-list attribute of user-label cluster fails

PSONALl, 2022-06-10:
#### Problem
In User Label Cluster Test case: ULABEL-2.4 step 1
 * expected behavior
   - TH writes LabelList attribute of the DUT with entry as following: "room":"bedroom 1","orientation":"east" " and returns success
 * actual behavior
    - TH returns an error: 0x01 (FAILURE)
    - read after write returns a empty list

 * system configuration
        -  DUT used: ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * log
        - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8877954/esp32.txt)
        - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8877956/th.txt)

bzbarsky-apple, 2022-06-10:
I believe this was fixed by https://github.com/project-chip/connectedhomeip/pull/18236, which post-dates the pre-sve SHA by a day or so.  @PSONALl are you in a position to verify that on tip?

PSONALl, 2022-06-14:
@bzbarsky-apple yes this issue is fixed now, so closing the issue :)

closed 2022-06-14 by PSONALl