project-chip/connectedhomeip #19450: [pre-sve] write on active-calendar-type attribute of Time Format Localization cluster fails

PSONALl, 2022-06-10:
#### Problem
 * expected behavior
     - TH writes 0 to ActiveCalendarType attribute and write request is successful with status code 0x00
 * actual behavior
     -  write fails with General error: 0x87 (CONSTRAINT_ERROR)

 * system configuration
          -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * logs
       -  esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8878233/esp32.txt)
       - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8878236/th.txt)

bzbarsky-apple, 2022-06-10:
I believe this was fixed by https://github.com/project-chip/connectedhomeip/pull/18236, which post-dates the pre-sve SHA by a day or so. @PSONALl are you in a position to verify that on tip?

PSONALl, 2022-06-14:
> I believe this was fixed by #18236, which post-dates the pre-sve SHA by a day or so. @PSONALl are you in a position to verify that on tip?

@bzbarsky-apple yes this issue is fixed now, so closing this issue

closed 2022-06-14 by PSONALl