project-chip/connectedhomeip #19451: [pre-sve] TC-DD-3.1 "PASE Session Established" logs are missing during commissioning

PSONALl, 2022-06-10:
#### Problem
In DD-3.1 step 2
 * expected behavior
       - during PASE session we must:
 > Check on DUT:
CHIP:SC: Received PBKDF param request
CHIP:SC: Peer assigned session ID 18450
CHIP:SC: Found MRP parameters in the message
CHIP:SC: Including MRP parameters in PBKDF param response
CHIP:EM: Piggybacking Ack for MessageCounter:850320430 on exchange: 63762r
CHIP:IN: Prepared unauthenticated message 0xaaaaad340560 to 0x0000000000000000 (0)  of type 0x21 and protocolId (0, 0) on exchange 63762r with MessageCounter:1341084110.
CHIP:IN: Sending unauthenticated msg 0xaaaaad340560 with MessageCounter:1341084110 to 0x0000000000000000 at monotonic time: 0000000000FBA380 msec
CHIP:SC: Sent PBKDF param response

> Check on TH:
Sent PBKDF param request
[1653471961.364996][30157:30162] CHIP:CTL: Setting wifi credentials from parameters
[1653471961.365051][30157:30162] CHIP:CTL: Setting attestation nonce to random value
[1653471961.365166][30157:30162] CHIP:CTL: Setting CSR nonce to random value
[1653471961.365285][30157:30162] CHIP:CTL: Commission called for node ID 0x0000000000000001
[1653471961.365797][30157:30163] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1653471961.366157][30157:30160] CHIP:DL: TRACE: Bus acquired for name C-75cd
[1653471961.382240][30157:30162] CHIP:DL: PlatformBlueZInit init success
[1653471961.393893][30157:30160] CHIP:BLE: BLE removing known devices.
[1653471961.397319][30157:30160] CHIP:BLE: BLE initiating scan.
[1653471961.415415][30157:30160] CHIP:BLE: Device 21:2D:EF:B9:9D:46 does not look like a CHIP device.

 * actual behavior
        - Commissioning is successful but these logs are missing
 
 * system configuration
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        

 *  logs:
        - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8878326/esp32.txt)
        - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8878327/th.txt)

bzbarsky-apple, 2022-06-10:
We can't force a DUT to have any particular logs, I would think.  The fact that the PASE session is established is indicated by the fact that we can send commands on it....

That said, on the chip-tool side we do in fact log:
```
[1654871911345] [91912:8954746] CHIP: [IN] New secure session created for device <FFFFFFFB00000000, 0>, LSID:6251 PSID:35443!
[1654871911345] [91912:8954746] CHIP: [CTL] Remote device completed SPAKE2+ handshake
[1654871911345] [91912:8954746] CHIP: [CTL] Stopping commissioning discovery over BLE
[1654871911345] [91912:8954746] CHIP: [BLE] CancelConnection
[1654871911345] [91912:8954746] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1654871911345] [91912:8954746] CHIP: [TOO] Pairing Success
```
to indicate successful PASE session establishment.  We could change that "Pairing Success" log to something that mentions PASE if that would be useful; that's what that log means.

@liamgonyea what would be most helpful here for this test plan?

PSONALl, 2022-06-13:
> We could change that "Pairing Success" log to something that mentions PASE if that would be useful; that's what that log means.

right, so should the test cases in chip-test-plan be updated accordingly?

bzbarsky-apple, 2022-06-14:
> right, so should the test cases in chip-test-plan be updated accordingly?

That's what I was asking @liamgonyea, yes.

liamgonyea, 2022-06-14:
@bzbarsky-apple I think the logs how they are in `chip-tool` saying "Pairing Success" are more clear that the pairing has successfully occurred altogether so we can keep those. It would be nice to add a line in with something along the lines of "PASE Establishment Successful" so that it is explicitly clear that the PASE process is complete

@PSONALl Looking at the TH=chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8878327/th.txt) logs, I do see the desired TH logs to indicate PASE establishment. The DUT logs on [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8878326/esp32.txt) are different though than on RasPi.

```
[1654672128.824504][3464:3469] CHIP:SC: Received PBKDF param response
[1654672128.824595][3464:3469] CHIP:SC: Peer assigned session ID 25170
[1654672128.824669][3464:3469] CHIP:SC: Found MRP parameters in the message
[1654672128.835752][3464:3469] CHIP:IN: Prepared unauthenticated message 0xffffb49e38b8 to 0x0000000000000000 (0)  of type 0x22 and protocolId (0, 0) on exchange 40280i with MessageCounter:250266660.
[1654672128.835812][3464:3469] CHIP:IN: Sending unauthenticated msg 0xffffb49e38b8 with MessageCounter:250266660 to 0x0000000000000000 at monotonic time: 0000000000860662 msec
[1654672128.835925][3464:3469] CHIP:SC: Sent spake2p msg1
[1654672128.920382][3464:3469] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1654672130.579460][3464:3467] CHIP:DL: Indication received, conn = 0xffffac0313c0
[1654672130.579863][3464:3469] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1654672130.580074][3464:3469] CHIP:EM: Received message of type 0x23 with protocolId (0, 0) and MessageCounter:3325877866 on exchange 40280i
[1654672130.580152][3464:3469] CHIP:EM: Found matching exchange: 40280i, Delegate: 0xffffa0000bb0
[1654672130.580241][3464:3469] CHIP:SC: Received spake2p msg2
[1654672130.583096][3464:3469] CHIP:IN: Prepared unauthenticated message 0xffffb49e39f8 to 0x0000000000000000 (0)  of type 0x24 and protocolId (0, 0) on exchange 40280i with MessageCounter:250266661.
[1654672130.583187][3464:3469] CHIP:IN: Sending unauthenticated msg 0xffffb49e39f8 with MessageCounter:250266661 to 0x0000000000000000 at monotonic time: 0000000000860D36 msec
[1654672130.583358][3464:3469] CHIP:SC: Sent spake2p msg3
[1654672130.675138][3464:3469] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1654672130.725068][3464:3467] CHIP:DL: Indication received, conn = 0xffffac0313c0
[1654672130.725401][3464:3469] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1654672130.725639][3464:3469] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:3325877867 on exchange 40280i
[1654672130.725746][3464:3469] CHIP:EM: Found matching exchange: 40280i, Delegate: 0xffffa0000bb0
[1654672130.725947][3464:3469] CHIP:IN: SecureSession Active 0xaaaaefd8f2b0 Type:1 LSID:30406
[1654672130.725997][3464:3469] CHIP:IN: New secure session created for device <FFFFFFFB00000000, 0>, LSID:30406 PSID:25170!
[1654672130.726099][3464:3469] CHIP:CTL: Remote device completed SPAKE2+ handshake
[1654672130.726145][3464:3469] CHIP:TOO: Pairing Success
```

closed 2022-06-16 by woody-apple