project-chip/connectedhomeip #19452: [pre-sve] DGSW-2.2 Implementation for event handler of Software Diagnostics events is missing

PSONALl, 2022-06-10:
#### Problem
In Test case DGSW-2
 * expected behavior
       - DUT sends an event report to TH. TH reads a list of SoftwareFault struct from DUT.
 * actual behavior
       - No Event Report is sent

 * system configuration
           -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
 *  logs
       - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8878444/esp32.txt)
       - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8878449/th.txt)

 #### Proposed Solution
 -  Add code to handle software events

franck-apple, 2022-06-10:
Bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

andy31415, 2022-06-13:
SoftwareFault event conformance is optional.

While this is useful for testing, I am unsure if this should block cert: event generation is up to the individual device implementation, software fault event generation is optional.

woody-apple, 2022-06-13:
Cert Blocker Review: Tagging @franck-apple to follow up with Sankar, to see if there's a test plan update/review needed.

sankarselvam, 2022-06-14:
@PSONALl  I checked with QA team, they used the following command to generate the event & then read the event. Can you please try and update your observation?

1. get the pid of the apps you are running by  "ps  -aef|grep all-clusters-app"
2.  Then kill the app using the following signal  "sudo kill -SIGUSR1 <pid from above command>"
3. Then read the event. 

It worked fine for us

PSONALl, 2022-06-14:
@sankarselvam tried this but I am not getting any Event Report when I read the event
this is because OnSoftwareFaultDetect is never called

sankarselvam, 2022-06-14:
OK, Seems like its a platform specific issue. We tried in RPI & we could able to see the called in DUT. Attached logs for re
[[TC-DGSW-2.2]TH.txt](https://github.com/project-chip/connectedhomeip/files/8899352/TC-DGSW-2.2.TH.txt)
[[TC-DGSW-2.2]DUT.txt](https://github.com/project-chip/connectedhomeip/files/8899353/TC-DGSW-2.2.DUT.txt)

PSONALl, 2022-06-14:
Yes @sankarselvam I am working to fix this issue on the esp32 side

closed 2023-05-08 by stale[bot]