project-chip/connectedhomeip #19455: [pre-sve] After bringing DUT into a state so that StateValue is TRUE, the value is not reflected in state attribute of Boolean State Cluster

PSONALl, 2022-06-10:
#### Problem
In TC-BOOL-2.2 in step 4d and 4e:
 * expected behavior
      - After manually bringing DUT into a state so that StateValue is TRUE when we read state attribute it should be true
 * actual behavior
      - there is no change in state

 * system configuration
          -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * logs:
       - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8878608/esp32.txt) 
       - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8878611/th.txt)

bzbarsky-apple, 2022-06-10:
@PSONALl What steps were used to "manually bring DUT into a state so that StateValue is TRUE"?  I am guessing those steps are not right, but it's hard to say without knowing what they were.

franck-apple, 2022-06-10:
Bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

andy31415, 2022-06-10:
assigning to PSONALI for adding addiontal info: please describe all steps to reproduce the issue. Specifically here we are missing at least the how to "manually bringing DUT into a state so that StateValue is TRUE "

andy31415, 2022-06-10:
Also please add how the `read attribute` is done (what attribute, what command, what endpoint...).

PSONALl, 2022-06-14:
@andy31415 for bringing DUT to true state I used cli option on DUT side to set attribute to true and reading the attribute with this command: `./chip-tool booleanstate read state-value 1 1`

bzbarsky-apple, 2022-06-14:
>  I used cli option on DUT side to set attribute to true 

Could you please describe the _exact_ commands you ran so that someone else, who is possibly not familiar with the setup, can try to reproduce your steps?

Because again, the working assumption is that this is where the issue is, but it's really hard to tell with the provided info.

vivien-apple, 2022-06-14:
Looking at the test plan, those steps are gated behind a PICS code: `M.ManuallyControlled` which is defined as: 
```Can the StateValue attribute be changed by manual operation at the DUT?```

The only place where I can see code  referencing `StateValue`is in the all-clusters-app (and the minimal version too) at https://github.com/project-chip/connectedhomeip/blob/f188fd9625f171a7d32a4b8e4a00d6abf60af891/examples/all-clusters-minimal-app/esp32/main/DeviceWithDisplay.cpp#L538

But I don't see any code to toggle it...

PSONALl, 2022-06-17:
> But I don't see any code to toggle it...

I had added a cli option to set an attribute to true on my own let me check that patch again and I will update here soon

PSONALl, 2022-06-17:
Tried changing state through CLI on DUT side on pre-sve week 2 SHA and this test case worked for me, so closing this issue

closed 2022-06-17 by PSONALl