project-chip/connectedhomeip #19456: [pre-sve] We cant set UserCanConsent option at run time so TC-SU-2.3 fails

PSONALl, 2022-06-10:
#### Problem
 * expected behavior
       - DUT should obtain user consent before transferring the image.
 * actual behavior
       - there is no option to obtain user consent at run-time

 * system configuration
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        

 * logs:
         - chip-tool : [chip-tool-4.txt](https://github.com/project-chip/connectedhomeip/files/8878832/chip-tool-4.txt)
         - ota-requestor logs:
[otar-1.txt](https://github.com/project-chip/connectedhomeip/files/8878837/otar-1.txt)
[otar-2.txt](https://github.com/project-chip/connectedhomeip/files/8878838/otar-2.txt)
[otar-4.txt](https://github.com/project-chip/connectedhomeip/files/8878839/otar-4.txt)
[otar-6.txt](https://github.com/project-chip/connectedhomeip/files/8878840/otar-6.txt)
          - corresponding ota-provider logs:
[otap-2.txt](https://github.com/project-chip/connectedhomeip/files/8878843/otap-2.txt)
[otap-6.txt](https://github.com/project-chip/connectedhomeip/files/8878847/otap-6.txt)

franck-apple, 2022-06-10:
Bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

woody-apple, 2022-06-13:
Cert Blocker Review: tagging @franck-apple to notify ESP

dhrishi, 2022-06-14:
@shubhamdp @jadhavrohit924 PTAL

PR #19677 (2022-06-16): [ESP32] Added cli to set requestoreCanConsent and UserConsentState at run-time
#### Problem
* There is no option for ota requestor to set requestorCanConsent and userConsentState.
* Fixes #19456

#### Change overview
Added cli option for ota requestor to set requestorCanConsent and userConsentState.

#### Testing
* Manually verified queryImageCommand and queryImageResponse.