project-chip/connectedhomeip #19458: [pre-sve] There is no option on OTA-Provider to change and populate action field in runTime

PSONALl, 2022-06-10:
#### Problem
In TC-SU-3.4 step 
 * expected behavior
      - In step 1 and step 2 we must be able to set Action field to values like Proceed or AwaitNextAction in run time and Verify if the action field is  populated correctly.

 * actual behavior
    -  action field is not populated correctly and always shows value 2

 * system configuration
         -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 *  logs with step numbers:
      - chip-tool logs:
[th-5.txt](https://github.com/project-chip/connectedhomeip/files/8878896/th-5.txt)
[th-7.txt](https://github.com/project-chip/connectedhomeip/files/8878897/th-7.txt)
      - ota-requestor-logs:      
[otar-1-2.txt](https://github.com/project-chip/connectedhomeip/files/8878900/otar-1-2.txt)
[otar-7.txt](https://github.com/project-chip/connectedhomeip/files/8878901/otar-7.txt)
      - ota-provider-logs:
[otap-1-2.txt](https://github.com/project-chip/connectedhomeip/files/8878925/otap-1-2.txt)
[otap-5.txt](https://github.com/project-chip/connectedhomeip/files/8878927/otap-5.txt)
[otap-4.txt](https://github.com/project-chip/connectedhomeip/files/8878928/otap-4.txt)
[otap-7.txt](https://github.com/project-chip/connectedhomeip/files/8878929/otap-7.txt)

franck-apple, 2022-06-10:
Bumping this issue to cert blocker as it's a request cert blocker discovered during pre-SVE tests.

woody-apple, 2022-06-13:
Cert Blocker Review: Tagging @franck-apple to notify ESP

isiu-apple, 2022-06-13:
See https://github.com/project-chip/connectedhomeip/pull/15208 for example Linux implementation.

dhrishi, 2022-06-14:
@shubhamdp @jadhavrohit924 PTAL

PR #19552 (2022-06-15): [ESP32] Added cli for change and populate action field in runTime
#### Problem
* There is no option on OTA-Provider to change and populate action field in runTime for ESP32
* Fixes #19458

#### Change overview
Added cli for change and populate action field in runTime for ESP32

#### Testing
* Manually verified applyUpdateResponce.
closed 2022-06-15 by andy31415