project-chip/connectedhomeip #19471: Objective-C Framework: We need a way to cancel subscriptions for specific attributes.

krypton36, 2022-06-10:
#### Problem
It currently is possible to subscribe to attributes in the Objective-C framework, but currently there is not way to cancel the subscription. We need an API that will allow for cancellation of the targeted subscription. 

Additional Suggestion: Add an additional API to cancel all subscriptions. For example, a user subscripts to attribute A, B, C, D and wants only E would do the following `cancelAll -> subscribe(E)` instead of `cancel(A) -> cancel(B) -> cancel(C) -> cancel(D) -> subscribe(E)`

 #### Proposed Solution
See above.

bzbarsky-apple, 2022-06-10:
The specific-attribute APIs currently use SDK APIs that do not allow canceling the subscription....

bzbarsky-apple, 2022-12-17:
@krypton36 Is this still an issue now that controller shutdown shuts down all subscriptions?

krypton36, 2023-01-03:
This is no longer an issue. closing.

closed 2023-01-03 by krypton36