project-chip/connectedhomeip #19481: [nRF52840-DK] low power sleep command on all-clusters-app crashes

dwgaul, 2022-06-10:
Ran the following command against the all-clusters-app on my nordic board:

```
./chip-tool lowpower sleep 0x0000000000003bfa 1
```

Seeing the following lines in mini com:

```
I: 10905 [IN]CASE Server enabling CASE session setups
I: 10914 [EM]Received message of type 0x8 with protocolId (0, 1) and MessageCounter:16126493 on exchange 15071r
D: 10925 [EM]Handling via exchange: 15071r, Delegate: 0x2000755c
D: 10932 [DMG]Received command for Endpoint=1 Cluster=0x0000_0508 Command=0x0000_0000
I: 10939 [ZCL]LowPower has no delegate set for endpoint:1
E: 10944 [ZCL]emberAfLowPowerClusterSleepCallback error: Error CHIP:0x00000003
D: 10951 [DMG]ICR moving to [ Preparing]
D: 10955 [DMG]ICR moving to [AddingComm]
D: 10959 [DMG]ICR moving to [AddedComma]
E: ***** USAGE FAULT *****
E:   Attempt to execute undefined instruction
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x2000b160
E: r3/a4:  0x20029800 r12/ip:  0x00000000 r14/lr:  0x0009e4af
E:  xpsr:  0x41000000
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x000032eb
E: s[ 4]:  0x00000000  s[ 5]:  0xffffffff  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0xffffffff  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0xffffffff  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00020e74
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2000b160 (CHIP)
E: Halting system
```

Not sure how critical this is, as we are testing media cluster tests against a nordic board mainly to cover thread transport.

bzbarsky-apple, 2022-06-11:
Code is forgetting to actually return on error and hence ending up with a null-deref.  PR coming up.

bzbarsky-apple, 2022-06-11:
With https://github.com/project-chip/connectedhomeip/pull/19494 the command will respond with Failure, but not crash.

closed 2022-06-14 by bzbarsky-apple