project-chip/connectedhomeip #19484: Too long TTL for SRP Update of commissionable node

kpark-apple, 2022-06-10:
#### Problem
It seems that Thread commissionable node SRP is updated using default TTL (1 hour I think).
Such a long TTL could cause an issue in next pairing.
Let's say a Thread node opened commissioning window in the Thread network.
It will update SRP with the long TTL.
And then, if user removes the node, power cycles it and attempt to commission the node again, the commissioner may find the matching record from DNS-SD first and attempts to commission it over IP and only after it fails with timeout, it would attempt to commission the node over BLE, which would be a bad user experience as user might think it's not working and give up beforehand.

 #### Proposed Solution
Reduce TTL of commissionable node SRP Update.

bzbarsky-apple, 2022-06-11:
@kpark-apple Is this basically the same as https://github.com/project-chip/connectedhomeip/issues/16858 ?

kpark-apple, 2022-06-11:
> @kpark-apple Is this basically the same as #16858 ?

Yes, it is. Thank you. Closing this one.

closed 2022-06-11 by kpark-apple