project-chip/connectedhomeip #19490: Scan network command throws (UNSUPPORTED_COMMAND) error with darwin-framework-tool

phonnakasturi-apple, 2022-06-11:
With SHA 5d8599d195a93dfe5473f64e4f888c322e5df1d2
 
**Test case [TC-DM-4.20] [WiFi] Verification for commands check [DUT-Commissioner]**

The test step where DUT sends the ScanNetwork command to the TH throws  (UNSUPPORTED_COMMAND) error with darwin-framework-tool

Attaching the logs;

[TC-DM-4.20_logs.txt](https://github.com/project-chip/connectedhomeip/files/8882854/TC-DM-4.20_logs.txt)

bzbarsky-apple, 2022-06-11:
@phonnakasturi-apple What is the server app involved?  Please always include the following in issues:

* The SHA (this issue has that)
* The client app (this issue has that)
* The serve app and what platform it's running on (missing)
* Logs for both sides (has client log, but not server log).

bzbarsky-apple, 2022-06-11:
If I had to guess: the server is something like all-clusters-app running on MacOS...

Our of curiosity, @phonnakasturi-apple, what does this return:
```
darwin-framework-tool networkcommissioning read accepted-command-list 1 0
```
?  I expect it comes back empty....

phonnakasturi-apple, 2022-06-13:
@bzbarsky-apple The server app is all-clusters-app and its running on raspi (ubuntu)

And the logs I have attached from both macOS and from raspi

For darwin-framework-tool networkcommissioning read accepted-command-list 1 0 ->

 sudo ./darwin-framework-tool networkcommissioning read accepted-command-list 1 0
[1655138441043] [67386:1184976] CHIP: [TOO] Running Command
[1655138441045] [67386:1184976] CHIP: [DMG] AccessControl: initializing
[1655138441045] [67386:1184976] CHIP: [DL] _Init
[1655138441045] [67386:1184976] CHIP: [IN] UDP::Init bind&listen port=5541
[1655138441045] [67386:1184976] CHIP: [IN] UDP::Init bound to port=5541
[1655138441045] [67386:1184976] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1655138441045] [67386:1184976] CHIP: [IN] TransportMgr initialized
[1655138441045] [67386:1184976] CHIP: [FP] Init fabric pairing table with server storage
2022-06-13 09:40:41.045 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: g/lkgt
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: g/lkgt, value {length = 8, bytes = 0x152600eafd342a18}
[1655138441047] [67386:1184976] CHIP: [TS] Last Known Good Time: 2022-06-09T18:52:26
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: g/fidx
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: g/fidx, value {length = 14, bytes = 0x1524000436010401040204031818}
[1655138441047] [67386:1184976] CHIP: [FP] Loading from storage for fabric index 0x1
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/1/r
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/1/r, value {length = 251, bytes = 0x15300101 00240201 37032714 8ab261d5 ... 3f537b95 de6ae518 }
2022-06-13 09:40:41.047 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/1/i
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/1/i, value (null)
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/1/n
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/1/n, value {length = 258, bytes = 0x15300101 01240201 37032714 8ab261d5 ... 28bbb8a8 22a7d918 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/1/o
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/1/o, value {length = 105, bytes = 0x15240001 30016104 a605c520 f77e7086 ... 9d333f14 89abcb18 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/1/m
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/1/m, value {length = 9, bytes = 0x152500f1ff2c010018}
[1655138441048] [67386:1184976] CHIP: [FP] Loading from storage for fabric index 0x2
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/2/r
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/2/r, value {length = 251, bytes = 0x15300101 00240201 37032714 dbc43307 ... f2840555 49609118 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/2/i
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/2/i, value (null)
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/2/n
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/2/n, value {length = 258, bytes = 0x15300101 01240201 37032714 dbc43307 ... 338712cc 462f1118 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/2/o
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/2/o, value {length = 105, bytes = 0x15240001 30016104 1d335c5a ca986791 ... e696b6ea 5e500918 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/2/m
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/2/m, value {length = 9, bytes = 0x152500f1ff2c010018}
[1655138441048] [67386:1184976] CHIP: [FP] Loading from storage for fabric index 0x3
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/3/r
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/3/r, value {length = 251, bytes = 0x15300101 00240201 37032714 f6505891 ... 3695f168 db040b18 }
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] PersistentStorageDelegate Sync Get Value for Key: f/3/i
2022-06-13 09:40:41.048 darwin-framework-tool[67386:1184976] CHIPPersistentStorageDelegate Get Value for Key: f/3/i, value (null)
[... cut ...]
[1655138441322] [67386:1184978] CHIP: [DMG] IM WH moving to [Uninitialized]
[1655138441322] [67386:1184978] CHIP: [DMG] IM WH moving to [Uninitialized]
[1655138441322] [67386:1184978] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1655138441322] [67386:1184978] CHIP: [BLE] CancelConnection
[1655138441322] [67386:1184978] CHIP: [DL] Inet Layer shutdown
[1655138441322] [67386:1184978] CHIP: [DL] BLE shutdown
[1655138441322] [67386:1184978] CHIP: [DL] System Layer shutdown
priyankahonnakasturi@Priyankas-MacBook-Pro debug %

bzbarsky-apple, 2022-06-13:
> The server app is all-clusters-app and its running on raspi (ubuntu)

@phonnakasturi-apple Compiled how?  It looks like it does not have wi-fi support compiled in.  Was this commissioned over BLE and joined a wi-fi network, or was it commissioned via on-network commissioning?

phonnakasturi-apple, 2022-06-13:
@bzbarsky-apple 
all-clusters-app was complied using the following steps..
```

cd ~/connectedhomeip
export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
source scripts/activate.sh
scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ out/all-clusters-app chip_inet_config_enable_ipv4=false

cd out/all-clusters-app
./chip-all-clusters-app 
```

bzbarsky-apple, 2022-06-13:
> scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ out/all-clusters-app chip_inet_config_enable_ipv4=false

Thanks.

@yufengwangca Why is this ending up without a working network commissioning cluster?  I'd expect `chip_enable_wifi` to be true on Linux by default, and then `chip_device_config_enable_wpa` should end up true, etc.  And similarly for thread...

woody-apple, 2022-06-14:
Assigning to @kean-apple to follow up.

krypton36, 2022-06-15:
@phonnakasturi-apple I was able to verify your issue here. I left out the `--wifi` parameter when running the `all-clusters-app` on RaspberryPi and I got UNSUPPORTED_COMMAND. I re-ran the `all-cluster-app` with `--wifi` and the command was successful.

bzbarsky-apple, 2022-06-15:
In that case that all sounds correct.  If `--wifi` is not passed, the app pretends to not know anything about wifi, and does that in a spec-compliant manner.

closed 2022-06-15 by bzbarsky-apple