project-chip/connectedhomeip #19499: Commissioning over BLE from macOS Venture for ESP32C3 not working

likidu, 2022-06-12:
#### Problem
I was trying to test with the example on ESP32C3 following this README: https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-minimal-app/esp32#commissioning-and-cluster-control

I can build and run the application okay. When I trying to use `chip-tool` to commission a device, it runs without errors but obviously the board does not connect to the WiFi.

I tested under macOS Venture beta 1. Don't know if that causes the problem with BLE.

Here is the output from chip-tool:
```txt
❯ ./out/debug/chip-tool pairing ble-wifi 19860206 S22Plus yjy64161551 20202021 3840

[1655074278019] [70038:288771] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1655074278039] [70038:288771] CHIP: [DL] _Init
[1655074278039] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=0
[1655074278039] [70038:288771] CHIP: [IN] UDP::Init bound to port=58709
[1655074278039] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=0
[1655074278039] [70038:288771] CHIP: [IN] UDP::Init bound to port=50265
[1655074278039] [70038:288771] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1655074278039] [70038:288771] CHIP: [IN] TransportMgr initialized
[1655074278039] [70038:288771] CHIP: [FP] Initializing FabricTable from persistent storage
[1655074278040] [70038:288771] CHIP: [TS] Last Known Good Time: 2022-06-12T14:46:01
[1655074278040] [70038:288771] CHIP: [FP] Loading from storage for fabric index 0x1
[1655074278041] [70038:288771] CHIP: [FP] Loading from storage for fabric index 0x2
[1655074278041] [70038:288771] CHIP: [FP] Loading from storage for fabric index 0x3
[1655074278043] [70038:288771] CHIP: [ZCL] Using ZAP configuration...
[1655074278043] [70038:288771] CHIP: [CTL] System State Initialized...
[1655074278043] [70038:288771] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1655074278044] [70038:288771] CHIP: [CTL] Setting attestation nonce to random value
[1655074278044] [70038:288771] CHIP: [CTL] Setting CSR nonce to random value
[1655074278044] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278044] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278049] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278049] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278049] [70038:288771] CHIP: [IN] TransportMgr initialized
[1655074278049] [70038:288771] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1655074278049] [70038:288771] CHIP: [CTL] Setting attestation nonce to random value
[1655074278049] [70038:288771] CHIP: [CTL] Setting CSR nonce to random value
[1655074278051] [70038:288771] CHIP: [CTL] Generating NOC
[1655074278052] [70038:288771] CHIP: [FP] Validating NOC chain
[1655074278052] [70038:288771] CHIP: [FP] NOC chain validation successful
[1655074278052] [70038:288771] CHIP: [FP] Added new fabric at index: 0x1, Initialized: 1
[1655074278052] [70038:288771] CHIP: [FP] Assigned compressed fabric ID: 0x4C7DF36C1E84C7BE, node ID: 0x000000000001B669
[1655074278055] [70038:288771] CHIP: [FP] Fabric (0x1) persisted to storage. Calling OnFabricPersistedToStorage
[1655074278055] [70038:288771] CHIP: [TS] Last Known Good Time: 2022-06-12T14:46:01
[1655074278055] [70038:288771] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1655074278055] [70038:288771] CHIP: [TS] Retaining current Last Known Good Time
[1655074278055] [70038:288771] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x4C7DF36C1E84C7BE
[1655074278055] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278055] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278055] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278055] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278055] [70038:288771] CHIP: [IN] TransportMgr initialized
[1655074278055] [70038:288771] CHIP: [CTL] Stopping commissioning discovery over DNS-SD
[1655074278055] [70038:288771] CHIP: [CTL] Setting attestation nonce to random value
[1655074278055] [70038:288771] CHIP: [CTL] Setting CSR nonce to random value
[1655074278056] [70038:288771] CHIP: [CTL] Generating NOC
[1655074278056] [70038:288771] CHIP: [FP] Validating NOC chain
[1655074278057] [70038:288771] CHIP: [FP] NOC chain validation successful
[1655074278057] [70038:288771] CHIP: [FP] Added new fabric at index: 0x2, Initialized: 1
[1655074278057] [70038:288771] CHIP: [FP] Assigned compressed fabric ID: 0x2BE06ADA22D27668, node ID: 0x000000000001B669
[1655074278061] [70038:288771] CHIP: [FP] Fabric (0x2) persisted to storage. Calling OnFabricPersistedToStorage
[1655074278061] [70038:288771] CHIP: [TS] Last Known Good Time: 2022-06-12T14:46:01
[1655074278061] [70038:288771] CHIP: [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1655074278061] [70038:288771] CHIP: [TS] Retaining current Last Known Good Time
[1655074278062] [70038:288771] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0x2BE06ADA22D27668
[1655074278062] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278062] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278062] [70038:288771] CHIP: [IN] UDP::Init bind&listen port=5550
[1655074278062] [70038:288771] CHIP: [IN] UDP::Init bound to port=5550
[1655074278062] [70038:288771] CHIP: [IN] TransportMgr initialized
[... 41 lines cut ...]
I (2106) chip[IN]: CASE Server enabling CASE session setups
I (2116) chip[SVR]: Joining Multicast groups
I (2116) chip[SVR]: Server Listening...
E (2126) chip[DL]: Long dispatch time: 535 ms, for event type 2
I (2136) app-devicecallbacks: Current free heap: 111016

I (2136) chip[DL]: WIFI_EVENT_STA_START
I (2146) phy: chan:1,max_power:80
I (2146) phy: chan:2,max_power:80
I (2146) phy: chan:3,max_power:80
I (2156) phy: chan:4,max_power:80
I (2156) phy: chan:5,max_power:80
I (2156) phy: chan:6,max_power:80
I (2166) phy: chan:7,max_power:80
I (2166) phy: chan:8,max_power:80
I (2166) phy: chan:9,max_power:80
I (2176) phy: chan:10,max_power:80
I (2176) phy: chan:11,max_power:80
I (2176) phy: chan:12,max_power:80
I (2196) phy: chan:13,max_power:80
I (2196) phy: chan:14,max_power:80
W (2206) wifi:Haven't to connect to a suitable AP now!
I (2206) chip[DL]: Done driving station state, nothing else to do...
I (2206) app-devicecallbacks: Current free heap: 111016
```

bzbarsky-apple, 2022-06-13:
> [1]    70038 abort      ./out/debug/chip-tool pairing ble-wifi 19860206 S22Plus yjy64161551 20202021

That is ... not "runs without error".  That's crashing before it manages to commission the device.

@likidu What SHA are you using?  Can you run chip-tool under lldb and debug the crash?

likidu, 2022-06-14:
@bzbarsky-apple, was not aware its aborted... The SHA is fdab0fd79feee007bc752bbf98c471d0246c4ebe.

I will do using `lldb` and see the crash.

likidu, 2022-06-14:
I just ran `lldb` with the same parameters and now it worked!
```bash
lldb -- ./out/debug/chip-tool pairing ble-wifi 19860206 S22Plus yjy64161551 20202021
```

Output from lldb:
```txt
[1655171554959] [1493:21511] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
2022-06-13 18:52:34.959178-0700 chip-tool[1493:21511] [all] 🔵 [1655171554959] [1493:21511] CHIP: [BLE] CancelConnection
[1655171554959] [1493:21511] CHIP: [BLE] CancelConnection
2022-06-13 18:52:34.959475-0700 chip-tool[1493:21511] [all] 🔴 [1655171554959] [1493:21511] CHIP: [DL] Inet Layer shutdown
[1655171554959] [1493:21511] CHIP: [DL] Inet Layer shutdown
2022-06-13 18:52:34.959493-0700 chip-tool[1493:21511] [all] 🔴 [1655171554959] [1493:21511] CHIP: [DL] BLE shutdown
[1655171554959] [1493:21511] CHIP: [DL] BLE shutdown
2022-06-13 18:52:34.959502-0700 chip-tool[1493:21511] [all] 🔴 [1655171554959] [1493:21511] CHIP: [DL] System Layer shutdown
[1655171554959] [1493:21511] CHIP: [DL] System Layer shutdown
2022-06-13 18:52:34.959531-0700 chip-tool[1493:21511] [all] 🟢 [1655171554959] [1493:21511] CHIP: [IN] SecureSession[0x1030dec80]: Released - Type:1 LSID:23590
[1655171554959] [1493:21511] CHIP: [IN] SecureSession[0x1030dec80]: Released - Type:1 LSID:23590
2022-06-13 18:52:34.959794-0700 chip-tool[1493:21648] [CoreBluetooth] XPC connection invalid
Process 1493 exited with status = 0 (0x00000000)
```

And macOS asked me the access to the Bluetooth so that's already an indication...

And I can see the device connects to the WiFi and successfully run further commands to turn on/off the LED.

Just don't know what's the differences there...

bzbarsky-apple, 2022-06-14:
OK, but running not under lldb is just crashing?

likidu, 2022-06-14:
@bzbarsky-apple Yes, at least that was I observed on my machine.

bzbarsky-apple, 2022-06-14:
@likidu And this is still reproducible?  That is, the fact that the permission was granted when running under a debugger does not affect what happens outside the debugger?

closed 2022-12-20 by stale[bot]