project-chip/connectedhomeip #19508: iOS  implicit instantiation of undefined template 'std::string'

mrcao2011, 2022-06-13:
#### Problem
<what's wrong or missing, please include any applicable:
run in real iphone is ok，but when run in  iOS simulator，have a error:
/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator15.0.sdk/usr/include/c++/v1/bitset:983:16: implicit instantiation of undefined template 'std::string'
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-06-13:
@mrcao2011 Is this compiling the iOS CHIPTool application?  What SHA of the SDK are you using?  What XCode version?

mrcao2011, 2022-06-14:
> @mrcao2011 Is this compiling the iOS CHIPTool application? What SHA of the SDK are you using? What XCode version?

yes，compiling the  iOS CHIPTool application  in iOS simulator，use the TE9 tag project，xcode version is 13.0。The strange thing is that the real machine（my iPhone XR） is normal。

mrcao2011, 2022-06-14:
> bzbarsky-apple

yes，compiling the iOS CHIPTool application in iOS simulator，use the TE9 tag project，xcode version is 13.0。The strange thing is that the real machine（my iPhone XR） is normal。

bzbarsky-apple, 2022-06-14:
> xcode version is 13.0

I just tried that SHA in xcode 13.2 targeting the "iPhone 11" simulator, as follows:

1. `git checkout cfc3595`
2. `git submodule update --init`
3. `. ./scripts/bootstrap.sh`
4. Start Xcode
5. Select the "CHIP Framework" target.
6. Select the "iPhone 12" similator.
7. Build.
8. Select the "CHIP Tool App" target.
9. Select the "iPhone 12" simulator.
10. Build.

and the build succeeded.

It's worth checking whether updating XCode to 13.2 helps, perhaps?

mrcao2011, 2022-06-17:
> I just tried that SHA in xcode 13.2 targeting the "iPhone 11" simulator, as follows:
>
> 1. `git checkout cfc3595`
> 2. `git submodule update --init`
> 3. `. ./scripts/bootstrap.sh`
> 4. Start Xcode
> 5. [...]

i have save the issuse,updating XCode to 13.4.1 is ok

closed 2022-06-17 by mrcao2011