project-chip/connectedhomeip #19516: [pre-sve] reset-watermarks command of software diagnostic cluster fails

PSONALl, 2022-06-13:
#### Problem
In TC-DGSW-2.3
 * expected behavior
        - TH sends ResetWatermarks to DUT and DUT’s reception of this command, DUT will reset StackFreeMinimum of ThreadMetrics attribute to StackFreeCurrent and reset CurrentHeapHighWatermark attribute to CurrentHeapUsed attribute.
 * actual behavior
       - Reset watermark command fails General error: 0x01 (FAILURE)
 * system configuration
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
 *  logs:
        - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8889769/dgsw-2.3-onwards-esp32.txt)
        - chip tool: [th-dgsw-2.3.txt](https://github.com/project-chip/connectedhomeip/files/8889770/th-dgsw-2.3.txt)

bzbarsky-apple, 2022-06-13:
esp32 does not support the optional watermark feature.  In #19195 the feature map was updated accordingly.

bzbarsky-apple, 2022-06-13:
@PSONALl This looks like it's an exactly duplicate of https://github.com/project-chip/connectedhomeip/issues/19454, which you also filed.  Why file twice?

closed 2022-06-13 by bzbarsky-apple